fra589/grbl-Mega-5X #162: M9 huge execute delay and G4 milliseconds?

erosenst, 2020-12-06:
Hello found that executing M9 makes big delay to movement, approx 0.5s. Is it posible to speed up it somehow?

Problem is that ramps 1.4 board heater output is controlled by M8/M9 combo and I connected plasma torch startup to that output. When shutdown torch it freeze movement ~0.5s and that time plasma burns too much metal. It dosn't shutdown immediatly so it must keep in movement.

And G4 P1 sleeps one second, not one microsecond. Imposible to do short delays  what is needed to plasma pierce etc..

Any ideas, patch to code?

fra589, 2020-12-07:
Hi @erosenst,  
  
The M7, M8 & M9 commands are for the flooding control. I don't think it's the right thing for controlling plasma. I did not read again all the flooding management code, these are functions that normally do not require an instant response time.
You rather should look at the laser management side instead, this should look more like your needs.
You can reassign all the output pins in the cpu_map.h file, the documentation for doing it is in the wiki: https://github.com/fra589/grbl-Mega-5X/wiki/Pinout-mapping-in-cpu_map.h

The `G4P*.***` GCode command accept floating point number as the P parameter. The value of P must be multiple of 50ms, this is the timing delay increment defined in config.h (`DWELL_TIME_STEP` at line 488). So, you can use a value smaller than one second for G4 timing. 
  
@++;  
Gauthier.

erosenst, 2020-12-08:
Hello I changed pins in cpu_map.h and enabled laser mode in config.h. But no success. It immediatly fires plasma when say M3. Documentation says that it should start when movement start. M3/M4/M5 combo have similiar execute delay/pause as M8/9.

I think laser mode code dosn't work as it should/documentation says. Any ideas/patches?

Thank you

erosenst, 2020-12-08:
[image]

Right sides of holes are good example what delay dos..

fra589, 2020-12-09:
Hi @erosenst,

> enabled laser mode in config.h
Have you also set the $32 parameter to 1 to activate the laser mode?

@++;
Gauthier.

erosenst, 2020-12-10:
$32=1 set already and dosn't work. Greping LASER from sources maybe tell why..

erosenst, 2020-12-13:
Hello found broblem in laser mode. There was $30=12000 so spindle speeds I used dosn't enough to lit LED or make detectable voltages to output (that was suprise "hotbed").
 There is still delay in M3/M4/M5/M7/M8/M9 executing but no more delay in S value changes so now cutting can do without brakehole..
For correct functionality it should't do brake in M3/M4/M5 but brake in S or don't if $32=1. Am I correct?

Can I close ticket anyway?

closed 2020-12-14 by fra589