fra589/grbl-Mega-5X #163: Streaming

crkie, 2020-12-12:
Hi there
I'm writing a G-code streamer for 4 axes. I use the ok counting method for this as recommended. This works wonderfully with the original GRBL, even with over 1000 lines.
With GRBL-Mega-5x I have the problem that it does not go through.
Could it be that somehow too many "ok" are being sent?

Sincerely 
Rudi

fra589, 2020-12-13:
Hi @crkie,  
  
What do you mean by "it does not go through"?  
Sending "OK" algorithm is the same as in the original grbl-Mega version that I forked for making 5X...  
  
@++;  
Gauthier.

crkie, 2020-12-13:
Hello Gauthier,
If the sendealogarytm is the same then the problem is with my program.
I tested it again with normal XYZ and these programs run completely without problems.
My goal is a hot wire version with XYUV. These programs are not yet fully executed. I have to keep looking for the bug.
But you can move four axes at the same time with G1? Or do I have to pay attention to something special here?

Thanks
Rudi

erosenst, 2020-12-13:
Is there some error or what is status after it don't execute?

crkie, 2020-12-13:
I have now increased the time between the question marks. It works now. I think the answers were incomplete. The Arduino obviously needs more time for 4 axes. There are many short distances for foam cutting. The Arduino is probably reaching its limits.

Thank you

fra589, 2020-12-14:
Hi Rudi,  
  
In my own Grbl sender [cn5X++](https://github.com/fra589/cn5X), I use a 75ms "?" polling delay and it work well.  
I made many tests to find this value which I think is the  optimum.  
  
@++;  
Gauthier.

closed 2020-12-14 by fra589