fra589/grbl-Mega-5X #165: Clone axis --- my fault

pferlini63, 2020-12-26:
Dear all,
i kindly ask your help for solving a my problem.
I'm using a ramps + Grbl mega for an autobuild CNC.

My question is related to clone axis. i've already read all articles in github and other sites. 

I'm trying to clone axe "Y" without success. I'm sure that hardware is ok, but when i try to move, the ex-"A" axes is engaged (but stopped). 
Therefore if (using cn5x) trying to move "A" axis, all works as expected.

Can you suggest me where i can find my problem ?

Best regards

Paolo

P.S. here below the $$ output and the part of config.h involved

$$
$0=10
$1=254
$2=0
$3=0
$4=0
$5=0
$6=0
$10=1
$11=0.020
$12=0.002
$13=0
$20=1
$21=0
$22=1
$23=15
$24=100.000
$25=500.000
$26=250
$27=5.000
$30=25000
$31=550
$32=0
$100=200.000
$101=200.000
$102=200.000
$103=200.000
$104=200.000
$110=750.000
$111=750.000
$112=750.000
$113=750.000
$114=750.000
$120=50.000
$121=50.000
$122=50.000
$123=50.000
$124=50.000
$130=300.000
$131=300.000
$132=150.000
$133=300.000
$134=300.000
ok
--------------------------------

#ifdef DEFAULTS_RAMPS_BOARD
  // 4, 5 & 6 axis support only for RAMPS 1.4 (for the moment :-)...)
  #define N_AXIS 4            // Number of axes
  #define N_AXIS_LINEAR 3     // Number of linears axis
#else
  #define N_AXIS 3 // Number of axes = 3 if not DEFAULTS_RAMPS_BOARD
#endif

#define AXIS_1 0        // Axis indexing value. Must start with 0 and be continuous.
#define AXIS_1_NAME 'X' // Axis names must be in X, Y, Z, A, B, C, U, V & W.
#define AXIS_2 1
#define AXIS_2_NAME 'Y'
#define AXIS_3 2
#define AXIS_3_NAME 'Z'

#if N_AXIS <3
  #error "N_AXIS must be >= 3. N_AXIS < 3 is not implemented."
#endif
#if N_AXIS > 3
  #define AXIS_4 3
  #define AXIS_4_NAME 'Y' // Letter of axis number 4
#endif
#if N_AXIS > 4
  #define AXIS_5 4
  #define AXIS_5_NAME 'B' // Letter of axis number 5
#endif
#if N_AXIS > 5
  #define AXIS_6 5
  #define AXIS_6_NAME 'C' // Letter of axis number 6
#endif

fra589, 2020-12-27:
Hi @pferlini63,  
Can you also put the output of the $I command ?
@++;  
Gauthier.

pferlini63, 2020-12-27:
Hi Gauthier,
thanks for your interest .

here below the output of $i command

>$I
[VER:1.1p.20201203:]
[AXS:5:XYZAB]
[OPT:VNMGH,35,255,48]
ok

pferlini63, 2020-12-27:
Hi Gauthier,
thanks for your request.

I finally discover my fault.

The problem was caused by PlatformIO because he is copying the sources in a new directory structure, while i was modifying the original one.

So any changes or investigation was on wrong sources (or sources never compiled and uploaded).

Thanks a lot for giving me the right direction to move.

:-)

Paolo

closed 2020-12-27 by pferlini63

fra589, 2020-12-27:
Hi Paolo,  
Don't forget to reset Grbl's parameters with the $RST=* command when you change the axis number or definition.  
Without this, Grbl may have unpredictable behavior in this case.  
@++;  
Gauthier.

pferlini63, 2020-12-27:
Thanks for this suggestion.
I'll do this afternoon.

Paolo