fra589/grbl-Mega-5X #167: Pin for driving relay to turn spindle on

boxgame9, 2021-01-04:
Hi Gauthier 
I have a 5v relay which I would use to turn the spindle on.
J'ai un relais 5v que j'utiliserais pour allumer la broche.

Pin D4 (2560) is the spindle enable pin and I used one of the end stop connections for the 5v supply.  Although the relay only requires 76ma to trigger it I cannot find a pin that will supply this. 
La broche D4 (2560) est la broche d'activation de la broche et j'ai utilisé l'une des connexions de butée pour l'alimentation 5v.  Bien que le relais ne nécessite que 76 mA pour le déclencher, je ne trouve pas de broche qui le fournira.

I have a pwm enabled spindle control board. Can I connect pin D6 (2560) directly to this input or should it also go through a relay?
J'ai une carte de commande de broche activée par pwm.  Puis-je connecter la broche D6 (2560) directement à cette entrée ou doit-elle également passer par un relais?

Thanks in advance and a happy new year
Merci d'avance et bonne année

Lesli

fra589, 2021-01-04:
Hi Lesli,  
The D4 Arduino pin is located on the ATmega2560 port G, Bit 5.  
From [ATmega 2560 datasheet](https://ww1.microchip.com/downloads/en/devicedoc/atmel-2549-8-bit-avr-microcontroller-atmega640-1280-1281-2560-2561_datasheet.pdf), you can see:  
"_4.)The sum of all IOL, for ports E0-E7, G5 should not exceed 100mA._"  
On Port E3 (D5) there is spindle direction,  
So, if you don't need the spindle direction output or if this outpur is less than 24mA, you can connect your 76mA relay to the Arduino D4 without problem.  
With the same calculation for the D6 Arduino port (you need to use #define SPINDLE_PWM_ON_D6 in config.h for this):
D6 is on ATmega2560 port H bit 3,  
From [ATmega 2560 datasheet](https://ww1.microchip.com/downloads/en/devicedoc/atmel-2549-8-bit-avr-microcontroller-atmega640-1280-1281-2560-2561_datasheet.pdf):  
"_3.)The sum of all IOL, for ports G3-G4, B0-B7, H0-H7 should not exceed 200mA._"  
on the Port B, bit 4, there is Coolant flood output,  
on the Port H, bit 6 there is Coolant mist output,  
So, the total 200mA output power must be shared for those 3 outputs.  
You can connect your card directly to the D6 Arduino port if the total consumption of your card + the coolants outputs is less than 200 mA.  

Happy new year,  
@++;  
Gauthier.

boxgame9, 2021-01-05:
Hi Gauthier 
I have connected both a pwm board with motor and a hand operated pwm board with relay and both worked fine.
J'ai connecté à la fois une carte pwm avec moteur et une carte pwm manuelle avec relais et les deux ont bien fonctionné.

thanks for your help
 Merci de votre aide

I will now close this issue 
Je vais maintenant fermer ce numéro

Lesli

closed 2021-01-05 by boxgame9