fra589/grbl-Mega-5X #168: The y-axes do not rotate in the positive direction when cloned

valeckUDL, 2021-01-07:
Hello
I want to use a clone of the Y axis. I made edits to the config file and to the axis parameters. But the Y-axis motors rotate correctly only in the negative direction. When attempting to return in the positive direction, the motors are blocked. In addition, the X-axis motor is trying to turn. No alarm is displayed.

$0=10
$1=255
$2=0
$3=0
$4=0
$5=0
$6=0
$10=1
$11=0.020
$12=0.002
$13=0
$20=1
$21=0
$22=1
$23=0
$24=500.000
$25=2000.000
$26=250
$27=5.000
$30=12000
$31=0
$32=0
$100=1600.000
$101=1600.000
$102=1600.000
$103=1600.000
$110=9000.000
$111=9000.000
$112=300.000
$113=9000.000
$120=300.000
$121=300.000
$122=100.000
$123=300.000
$130=250.000
$131=470.000
$132=35.000
$133=470.000

[VER:1.1l.20190605:]
[AXS:4:XYZY]
[OPT:VNM,35,255,48]

#ifdef DEFAULTS_RAMPS_BOARD
  // 4, 5 & 6 axis support only for RAMPS 1.4 (for the moment :-)...)
  #define N_AXIS 4            // Number of axes
  #define N_AXIS_LINEAR 3     // Number of linears axis
#else
  #define N_AXIS 3 // Number of axes = 3 if not DEFAULTS_RAMPS_BOARD
#endif

#define AXIS_1 0        // Axis indexing value. Must start with 0 and be continuous.
#define AXIS_1_NAME 'X' // Axis names must be in X, Y, Z, A, B, C, U, V & W.
#define AXIS_2 1
#define AXIS_2_NAME 'Y'
#define AXIS_3 2
#define AXIS_3_NAME 'Z'

#if N_AXIS <3
  #error "N_AXIS must be >= 3. N_AXIS < 3 is not implemented."
#endif
#if N_AXIS > 3
  #define AXIS_4 3
  #define AXIS_4_NAME 'Y' // Letter of axis number 4
#endif
#if N_AXIS > 4
  #define AXIS_5 4
  #define AXIS_5_NAME 'B' // Letter of axis number 5
#endif

fra589, 2021-01-07:
Hi @valeckUDL,  
  
In your config.h, you must also change the number of linears axis: `#define N_AXIS_LINEAR 4`, since your 4th cloned axis is a linear axis.  
Unfortunately, I think it's not this configuration detail that's causing your problem.  
  
On the problem you describe, I see 2 possibilities:  
- A hardware problem (the D28 port of the Arduino board does not work correctly or the motor driver has a problem),  
- You have tested different axis configuration (number and definitions) and the Grbl configuration EEPROM is corrupted.  
  
In case of corruption of the EEPROM which stores the parameters of Grbl, the behavior can be completely unpredictable.  
In this case, the Grbl command: `$RST=*` makes it possible to reset factory defaults and to find normal operation.  
  
@++;  
Gauthier.  
P.S. Your grbl-Mega-5X version is not up to date, the latest up to date version is: 1.1q.20210103.

valeckUDL, 2021-01-07:
I followed your recommendations. Including downloaded the latest firmware. It all worked. 
Thank you very much))

fra589, 2021-01-07:
Your welcome :smiley:

closed 2021-01-07 by fra589