fra589/grbl-Mega-5X #170: Can we merge this in some way ?

jlirochon, 2021-01-08:
Hi Gauthier,

Happy new year, hope you are doing great !

Some time ago, @johnboiles forked grbl-Mega-5X and added the following features :
- UltiMachine RAMBo 1.4 boards support
- digipots support (used by RAMBo boards)
- A way to "square" a machine having 2 steppers for X + 2 steppers for Y. It uses 2 axes for X and 2 axes for Y. Each axis has its own endstop, and each endstop can have its own offset in $ettings.

He did this to support MPCNC machines, but I wonder if the changes could be reintegrated at least partially into grbl-Mega-5X ? Actually the code seems to be too specific to the MPCNC, but I can refactor into generic features matching your requirements if you are OK with the idea. The goal here is to reduce the burden of merging upstream changes + there is no will to drive a dedicated project.

You can see his work here : https://github.com/johnboiles/grbl-Mega-5X/commits/mpcnc-rambo-support

Let me know what you think. If you want we can talk about it in some Western Europe dialect ^^

johnboiles, 2021-01-08:
👋 👋 Happy to slice my fork up into discrete PRs once we get @fra589's high level approval to merge the features.

fra589, 2021-01-08:
Hi @jlirochon, Hi @johnboiles,
  
Happy new year to you too,  
Currently, I am on vacation in the island of Martinique.  
So I do not have enough time to study your proposal and how we can integrate John's work according to what I have already planned for the future and the evolutions of grbl-Mega-5X.  
I will look into this subject after my return in a few weeks, when time will give me a bit of a break.  
  
@++;  
Gauthier.

johnboiles, 2021-01-08:
Wow enjoy your vacation!!!

RaphaelDives, 2021-01-09:
> * A way to "square" a machine having 2 steppers for X + 2 steppers for Y. It uses 2 axes for X and 2 axes for Y. Each axis has its own endstop, and each endstop can have its own offset in $ettings.

I'm running grbl-mega-5x on my MPCNC (with rampps board) and I'm using the autosquare feature described above since the beginning - what is the difference to the original code? Ok, I don't know if it's possible to define offsets, but I never had the need of it. But I'm always looking for improvements! :)

Happy new (and better) year @all!
And @fra589 have a nice time and hopefully the weather is better than in Germany. ;)

Raphael

jlirochon, 2021-01-09:
@RaphaelDives the use case is that you put your endstops in a sensible position, but instead of tuning the machine by moving the endstops, you can tune by settings offsets. I am plotting the biggest rectangle I can, then by measuring diagonals of the rectangle I can tell if it's skewed or not. If it's skewed I can slightly increment the offset (say +0.20mm / +0.008in) on one well chosen endstop, and try again until it's perfect. I never tried the original flavor of grbl-Mega-5x so I can only imagine it's a bit more tedious to move the endstops mechanically, but maybe it's not ?

Other changes are dedicated to RAMBo boards, which I see as a slightly improved version of a RAMPS board with included stepper drivers.

RaphaelDives, 2021-02-17:
@jlirochon yes, that IS a tedious task, especially because moving one endstop also influences the position of the other... :D 
(But that problem will also be present with special offsets...)
Anyway, this feature could really be useful.
On the other hand... I did this only once. Till next time when I will disasamble my MPCNC to add new stuff... :D
Cheers
Raphael