fra589/grbl-Mega-5X #173: Z axis homing issue - X & dual Y fine though

jeparky, 2021-01-21:
Wanted to start off saying thank you to everyone that helps out on here. I’ve learned many things from other post. This is my first time asking for help since I’ve been able to figure everything else out so far with lots of reading and researching. Problem is I can’t figure out what is causing my issue with homing my Z axis. My switch is fine and UGS even shows all my switches work. If I set my Z axis somewhat close to the homing switch and do a home cycle it touches the switch but then keeps on pushing on it. If I move my Z axis away a good bit then run a home cycle it searches twice but only short travels (why is that?) and then goes on to home X and then my dual Y axis’s just fine. I also have video I posted on YouTube of both ways to show exactly what it’s doing. I get an Alarm 8 when it crashes (first video) but even if I adjust bounce delay delay and/or pull off distance that doesn’t fix it or even adjusting motor settings. I’m at a road block.

https://youtu.be/NryDcNb2ZR0

https://youtu.be/-6lNDNFQplM

[VER:1.1q.20210103:]
[AXS:4:XYZY]
[OPT:VNMGH,35,255,16]

Config.h
#ifdef DEFAULTS_RAMPS_BOARD
  // 4, 5 & 6 axis support only for RAMPS 1.4 (for the moment :-)...)
  #define N_AXIS 4            // Number of axes
  #define N_AXIS_LINEAR 3     // Number of linears axis
#else
  #define N_AXIS 3 // Number of axes = 3 if not DEFAULTS_RAMPS_BOARD
#endif

#if N_AXIS <3
  #error "N_AXIS must be >= 3. N_AXIS < 3 is not implemented."
#endif
#if N_AXIS > 3
  #define AXIS_4 3
  #define AXIS_4_NAME 'Y' // Letter of axis number 4
#endif

#if N_AXIS == 4 // 4 axis : homing
    #define HOMING_CYCLE_0 (1<<AXIS_3) // Home Z axis first to clear workspace.
    #define HOMING_CYCLE_2 (1<<AXIS_1) // Home X axis
    #define HOMING_CYCLE_3 ((1<<AXIS_2)|(1<<AXIS_4)) // Home Y axis
    // #define HOMING_CYCLE_1 (1<<AXIS_4) // Home 4th axis (A)

$0 = 10    (Step pulse time, microseconds)
$1 = 255    (Step idle delay, milliseconds)
$2 = 0    (Step pulse invert, mask)
$3 = 11    (Step direction invert, mask)
$4 = 0    (Invert step enable pin, boolean)
$5 = 0    (Invert limit pins, boolean)
$6 = 0    (Invert probe pin, boolean)
$10 = 1    (Status report options, mask)
$11 = 0.020    (Junction deviation, millimeters)
$12 = 0.002    (Arc tolerance, millimeters)
$13 = 0    (Report in inches, boolean)
$20 = 0    (Soft limits enable, boolean)
$21 = 0    (Hard limits enable, boolean)
$22 = 1    (Homing cycle enable, boolean)
$23 = 5    (Homing direction invert, mask)
$24 = 100.000    (Homing locate feed rate, mm/min)
$25 = 500.000    (Homing search seek rate, mm/min)
$26 = 25    (Homing switch debounce delay, milliseconds)
$27 = 5.000    (Homing switch pull-off distance, millimeters)
$30 = 12000    (Maximum spindle speed, RPM)
$31 = 550    (Minimum spindle speed, RPM)
$32 = 0    (Laser-mode enable, boolean)
$100 = 400.000    (X-axis travel resolution, step/mm)
$101 = 200.000    (Y-axis travel resolution, step/mm)
$102 = 200.000    (Z-axis travel resolution, step/mm)
$103 = 200.000   
$110 = 2500.000    (X-axis maximum rate, mm/min)
$111 = 2500.000    (Y-axis maximum rate, mm/min)
$112 = 1000.000    (Z-axis maximum rate, mm/min)
$113 = 2500.000   
$120 = 50.000    (X-axis acceleration, mm/sec^2)
$121 = 50.000    (Y-axis acceleration, mm/sec^2)
$122 = 50.000    (Z-axis acceleration, mm/sec^2)
$123 = 50.000   
$130 = 400.000    (X-axis maximum travel, millimeters)
$131 = 640.000    (Y-axis maximum travel, millimeters)
$132 = 140.000    (Z-axis maximum travel, millimeters)
$133 = 640.000

fra589, 2021-01-22:
Hello @jeparky,  
  
It look like you may have and hardware problem...  
On your videos, it seems than you have an hardware problem: Perhaps than the Arduino's D48 pin is not working correctly (D48 is the location of direction bit of the third axis defined in cpu_map.h if you doesn't have changed it). So, the Z axis can't change it's move direction.  
  
@++;  
Gauthier.

RaphaelDives, 2021-02-17:
Hey @jeparky,
I had a very similar problem - It could be the same in your case:

I had cables for each switch on my axis and to save cables to the Z axis, I ran both the motor wires and the switch in one cable... However with manual switch testing, every switch worked fine. But while homing, my Z-axis showed the same behaviour, while X and Y worked fine. ;) (I thougt I was just to stupid to configure my grbl correctly and lived with that...)

I understood the problem, when I changed my design and conveniently ran all the endstops with the motor wires. None of the axis would do a homing. Just two little moves and stop without a reason and miles away from the switch.
So in MY case I'm running my steppers at 24V (but I guess the same happens with 12V) and the switches - directly connected to the atmega operates - with 5V. I think 24V are very good in inducing current in my switch wires. At least enough, that the Atmega would trigger. 
The suggested solution for this problem is to support the internal pullups with external resistors and use an capacitor to stabilize the input (RC circuit) - none of that helped in my case, probably of the 24V... ;)

Now I'm running my endstops at 24V with relais for each. That never failed up to now. ;)
I'm using an 8-channel relais module with optocuplers(?).

I would just try to connect an extra endstop (with short cables) to the board and trigger it by hand while homing. I bet that will work. ;)

Good luck!
Cheers
Raphael

closed 2022-06-26 by fra589