fra589/grbl-Mega-5X #174: LCD ??

broncosis, 2021-01-21:
https://github.com/bdurbrow/grbl-Mega

I have a mash up of your stuff with this stuff up and running finally 
I need D8 for my  spindle control 
but really wanted the LCD option for my ramps setup 

your code base is fairly similar I don't think it would be to hard to merge

fra589, 2021-01-21:
It's not in my TODO list to add LCD option nor SD card option to grbl-Mega-5X.
I think I said many time why I don't want it because it probably will need breaking many of Grbl's optimization and performance.
If you want those functionalities and if you think as you say than it is not hard to do it, do it yourself in your own project.
For myself, I don't want it!

closed 2021-01-21 by fra589

erosenst, 2022-03-29:
Hello, I done it years ago. You find it googling "grbl-5xlt". Source is there closed issues somewhere..