fra589/grbl-Mega-5X #175: PCB

victor999, 2021-01-26:
First of all I want to thank you all for this excellent and very useful project.
I designed a PCB for this project and I want to share it - free. I am running CNC with dual X axis and everything works. The PCB designed for XYZXAB configuration with limit switches, so there are changes in config as well. What is the best way to share it? 

I want to share schematic and board files, created in Eagle. My board works with external drivers, like TB6600

jonotaegi, 2021-01-29:
Hi @victor999,

I am really interested in your Eagle design and config files. I'm building a CNC machine very similar to yours with a dual axis (Y in my case) with limit switches and external TB6600 drivers.

Can you share your files in a GitHub repository?

Thanks!

victor999, 2021-01-29:
I hope it's OK, I posting a link to my repository
https://github.com/victor999/grbl-mega-board

fra589, 2021-01-29:
Hi @victor999,  
  
Currently, I am reorganizing the code in order to clean the old 3 axis only code, as well as all the old specific configurations from the old Grbl on Arduino Uno.  
After that, it should be easy to set options for specific cards or machines by providing a specific config file and just add a `#define ` statement to point to this file.  
  
@++;  
Gauthier.

closed 2022-06-26 by fra589