fra589/grbl-Mega-5X #176: Grbl z-axis port as second y-axis?

zkreider, 2021-01-27:
Hi all,

I currently have a controller running grbl1.1h that allows for one motor per x,y,z axis. My laser table doesn't have a need for z axis control. Is there a way to use the spare z-axis port for a second y-axis motor? I'm having trouble finding a simple fix in the grbl source code. Any help would be greatly appreciated. Thanks!

taibhse-designs, 2021-01-29:
Are you able to wire the motors in parallel? You could use the one y port for both motors then.

fra589, 2021-01-29:
Hi @zkreider,  
  
grbl-Mega-5X run on Arduino Mega 2560 and it's able to clone any axis on any hardware output by simply affect the name of axis in the config.h file.
  
See : https://github.com/fra589/grbl-Mega-5X/issues/4#issuecomment-404512389 or search for grbl-mega-5x clone axis in Google for more info.  
  
@++;  
Gauthier.

closed 2021-02-15 by fra589