fra589/grbl-Mega-5X #177: Dual motor homing with 1 limit switch

d-portero, 2021-01-30:
I have two motors on my Y axis. I only have one limit switch on Y right now. When I home Y and the switch is enabled, only one motor stops moving. Is there a way to set the second motor to stop moving as well? 

[image]
[image]

RaphaelDives, 2021-01-30:
Hey,
I don’t know, if there is an educated way to achieve this, but you can just set the same port and the same pin for both axis. (I found this out accidentally... 😂)
Cheers
Raphael

fra589, 2021-01-30:
Hi @d-portero,  
  
As @RaphaelDives said (thanks Raphael :smile:), you also must change the port definition in the cpu_map.h file.  
On ATMega 2560, there are 8 pins per port (one pin for each bit) and the cpu_map.h file reflect this definition: First we must define on which port is connected a functionality, then, we define on which bit.  
So, you need to change like this:
```
  #define MAX_LIMIT_PORT_0 E
  #define MAX_LIMIT_PORT_1 J
  #define MAX_LIMIT_PORT_2 D
  #if N_AXIS > 3
    #define MAX_LIMIT_PORT_3 J
  #endif
```  
@++;  
Gauthier.

d-portero, 2021-01-30:
Oh I totally missed that! Problem fixed. Thanks guys!

closed 2021-01-30 by fra589