fra589/grbl-Mega-5X #178: Bluetooth issue

arlimbad222, 2021-02-10:
Hello 
Today I connect bluetooth module (hc-05) to my mega at 115200 build & connect to ugs but machine behavior was strange . I can run small gcode files but when it comes to large file with more complex profile machine was jerking usb was working perfect but bluetooth ah very slow motion . Anyone have a solution for this .

fra589, 2021-02-11:
Hi @arlimbad222,  
  
With so little information it's impossible to help you!

How do you have connected your bluetooth module? On which pins? Have you kept a dual Bluetooth + USB connection? Do you have added code to Grbl to use the Bluetooth module? In this case, which code?  
And so on.  
All information to know exactly what you have done to obtain the behavior you describe are needed.  
  
@++;  
Gauthier.

arlimbad222, 2021-02-11:
hrelo @fra589 

currently i am using only bluetooth connection only in TX RX pin of arduino mega connect laptop to that HC-05 bluetooth module and put that COMPORT in UGS connect fast but when i jog machine or run gcode machine behave strange .

i only run one thing at a time like when bluetooth connect i remove usb and when usb connect i remove bluetooth module.
i search something on google i thik its baud rate issue with grbl and bluetooth communication .

can you please help me i dont know hoe to increase baud rate of grbl .

fra589, 2021-02-11:
Hi @arlimbad222,  
  
What do you mean by "machine behave strange"?  
  
In the HC-05 datasheet (https://components101.com/sites/default/files/component_datasheet/HC-05%20Datasheet.pdf), I saw that the default baud rate of this module is 38400.
The default bauds rate of Grbl is 115200. In front of the 38400 baud rate of your module, may be it's your problem source.  
The complete serial configuration of Grbl is 115200 bauds, 8 data bits, 1 stop bit and no parity.  
You need to configure your module with AT command (AT+UART=...) to configure 115200 bauds, one stop bit and parity none.  
In Grbl, the serial speed is defined in config.h at line 48. I strongly advise you not to configure a speed higher than 115200 because this can lead to saturation of the processor and therefore Grbl errors which can have serious consequences on the position control of the motors.
  
@++;  
Gauthier.

arlimbad222, 2021-02-11:
Hi @fra589 

I do the same I changed the baud rate of bluetooth module by 115200 but that speed is not enough for communication is there any way to control cnc wirelessly. 

Thankyou for response .

fra589, 2021-02-13:
Hi @arlimbad222,  
  
> What do you mean by "machine behave strange"?  
  
You still haven't accurately described the behavior you are having trouble with.  
  
Regarding the communication speed of Grbl, I think you were wrong in saying that 115200 baud is not enough.  
  
Indeed, if we consider a GCode command of 30 characters, which is rather the size of the longest movement commands, 115200 bauds can send 480 commands per second!  
It seems to me much higher than the need for the operation of a CNC machine. Even with 38400 baud, we can send nearly 160 commands of 30 characters per second to Grbl, which seems more than enough.  
  
You announce that with a USB connection, Grbl works very well at the speed of 115200 bauds, and that you have problems (what are these problems exactly ???) with the Bluetooth connection, it is not a problem of operation from Grbl, but a communication problem.  
  
Communication problems usually occur when the speed is too high rather than too low. It is very likely that your Bluetooth module is not able to provide full duplex communication at 115200 baud without errors.  
So I advise you to try lower speeds to see if the behavior of the link becomes better.  
Start by trying the default speed of your Bluetooth module at 38400 baud (In Grbl, the serial speed is defined in config.h at line 48 and don't forget to issue a `$RST=*` Grbl command after flashing it), and if you still have problems, you can even try at 19200 to see the behavior evolution.  
  
@++;  
Gauthier.

arlimbad222, 2021-02-13:
hi @fra589 

I find the issue last night i think its not grbl but bluetooth module . like at 115200 usb perforated well but when it comes to bluetooth machine take a small delay to Perform every moves . I cange this line number 511 at config.h (#define BLOCK_BUFFER_SIZE 90  // Uncomment to override default in planner.h.)  & machine run smooth at starting and after some moves again same problem it start lagging movements etc...

I try at lower baud today & comment here results 

thank you for response

arlimbad222, 2021-02-13:
Here is that file that i run for testing & videos of difference between bluetooth and usb connection . i cant find proper solution for this thing .
[punch2.txt](https://github.com/fra589/grbl-Mega-5X/files/5975289/punch2.txt)

(bluetooth)
https://user-images.githubusercontent.com/34487708/107844866-38153700-6dfd-11eb-8626-389e7eb23085.mp4

(usb)
https://user-images.githubusercontent.com/34487708/107844887-609d3100-6dfd-11eb-8337-14b970d9991a.mp4

fra589, 2021-02-15:
Hi @arlimbad222,  
  
The videos you posted clearly show a communication problem unrelated to the speed of the serial link.  
Having bluetooth stuttering issues seems to be quite common and can be caused by many causes, in your case it could be radio interference, compatibility between your computer's bluetooth transmitter and the HC-05 module or other things?  
Your problem is not with the speed configuration or a malfunction of Grbl, and unfortunately I cannot help you further on this.  
  
@++;  
Gauthier.

arlimbad222, 2021-02-15:
Hi @fra589 

Thankyou for response i can understand that this problem is out of my head also but I'll find some solutions later I am working on it whenever I find anything I'll get back here. 

Thankyou again and again brother .

closed 2021-02-17 by fra589

arlimbad222, 2021-02-26:
Hi @fra589 

I found issue last night and the issue is power supply of bluetooth module and bluetooth module i attach photo of 2 bluetooth module and the left one is clone copy of hc-05 module so be careful when buy a module like hc-05 there is many company made copy of hc-05. now my machine work fine with bluetooth .
[image]

thankyou again and again brother .

fra589, 2021-02-26:
Hi @arlimbad222,  
  
Thank you for the feedback 😃   
  
@++;  
Gauthier.

arlimbad222, 2021-02-26:
@fra589 

Your welcome sir I almost give my week to find this issue so I comment it because maybe this issue help someone else in future .

arlimbad222, 2021-02-28:
Hi @fra589 

There is one small issue with convert communication , bluetooth to USB whenever I want to switch Between bluetooth to USB it requires to pullout tx rx pin of bluetooth is there any solution for this like multiple communication options ....

Sorry for this shit question.