fra589/grbl-Mega-5X #179: Feed axis half speed from F540 to F600 bugs

78Matteo78, 2021-02-16:
Hi
I did several speed tests and I noticed that from f540 to f600 in the development of the gcode the speed is half, other times the flow of data is blocked.
The problem also occurred in another cnc of a friend.
I wanted to know if it happens to you too and if you can fix it.
thanks greetings matteo.

fra589, 2021-02-17:
Hi @78Matteo78,  
  
I don't quite understand what's happening to you...  
The calculation of speeds and acceleration in Grbl is mathematical, and I can't hardly see that 540/2 = 600...  
It takes a lot more information to be able to analyze what could explain the abnormal behavior you are seeing.  
What is your exact version of Grbl-Mega-5X? how many axes? Are there cloned axes? => Result of the command $I  
What is the complete configuration of Grbl? => result of the $$ command  
What are the GCode commands that reproduce the phenomenon?  
Etc... Any other information that could help to find...  
  
@++;  
Gauthier.

78Matteo78, 2021-02-17:
my setup value

$$
< $0=10          (Sets time length per step. Minimum 3usec.)
< $1=25          (Sets a short hold delay when stopping to let dynamics settle before disabling steppers. Value 255 keeps motors enabled with no delay.)
< $2=0           (Inverts the step signal. Set axis bit to invert (00000ZYX).)
< $3=0           (Inverts the direction signal. Set axis bit to invert (00000ZYX).)
< $4=1           (Inverts the stepper driver enable pin signal.)
< $5=0           (Inverts the all of the limit input pins.)
< $6=0           (Inverts the probe input pin signal.)
< $10=1          (Alters data included in status reports.)
< $11=0.010      (Sets how fast Grbl travels through consecutive motions. Lower value slows it down.)
< $12=0.002      (Sets the G2 and G3 arc tracing accuracy based on radial error. Beware: A very small value may effect performance.)
< $13=0          (Enables inch units when returning any position and rate value that is not a settings value.)
< $20=0          (Enables soft limits checks within machine travel and sets alarm when exceeded. Requires homing.)
< $21=0          (Enables hard limits. Immediately halts motion and throws an alarm when switch is triggered.)
< $22=0          (Enables homing cycle. Requires limit switches on all axes.)
< $23=0          (Homing searches for a switch in the positive direction. Set axis bit (00000ZYX) to search in negative direction.)
< $24=25.000     (Feed rate to slowly engage limit switch to determine its location accurately.)
< $25=500.000    (Seek rate to quickly find the limit switch before the slower locating phase.)
< $26=250        (Sets a short delay between phases of homing cycle to let a switch debounce.)
< $27=1000.000   (Retract distance after triggering switch to disengage it. Homing will fail if switch isn't cleared.)
< $30=255        (Maximum spindle speed. Sets PWM to 100% duty cycle.)
< $31=0          (Minimum spindle speed. Sets PWM to 0.4% or lowest duty cycle.)
< $32=0          (Enables laser mode. Consecutive G1/2/3 commands will not halt when spindle speed is changed.)
< $100=400.000   (X-axis travel resolution in steps per millimeter.)
< $101=400.000   (Y-axis travel resolution in steps per millimeter.)
< $102=400.000   (Z-axis travel resolution in steps per millimeter.)
< $103=13.300    (A-axis travel resolution in steps per degre.)
< $104=12.500    (B-axis travel resolution in steps per degre.)
< $110=900.000   (X-axis maximum rate. Used as G0 rapid rate.)
< $111=900.000   (Y-axis maximum rate. Used as G0 rapid rate.)
< $112=500.000   (Z-axis maximum rate. Used as G0 rapid rate.)
< $113=900.000   (A-axis maximum rate. Used as G0 rapid rate.)
< $114=200.000   (B-axis maximum rate. Used as G0 rapid rate.)
< $120=20.000    (X-axis acceleration. Used for motion planning to not exceed motor torque and lose steps.)
< $121=20.000    (Y-axis acceleration. Used for motion planning to not exceed motor torque and lose steps.)
< $122=20.000    (Z-axis acceleration. Used for motion planning to not exceed motor torque and lose steps.)
< $123=100.000   (A-axis acceleration. Used for motion planning to not exceed motor torque and lose steps.)
< $124=20.000    (B-axis acceleration. Used for motion planning to not exceed motor torque and lose steps.)
< $130=20000.000 (Maximum X-axis travel distance from homing switch. Determines valid machine space for soft-limits and homing search distances.)
< $131=20000.000 (Maximum Y-axis travel distance from homing switch. Determines valid machine space for soft-limits and homing search distances.)
< $132=20000.000 (Maximum Z-axis travel distance from homing switch. Determines valid machine space for soft-limits and homing search distances.)
< $133=20000.000 (Maximum A-axis travel distance from homing switch. Determines valid machine space for soft-limits and homing search distances.)
< $134=20000.000 (Maximum B-axis travel distance from homing switch. Determines valid machine space for soft-limits and homing search distances.)
< ok

78Matteo78, 2021-02-17:
Hi

I created a square with a side of 150 mm with speed f600 and in theory it should take 1 minute, but the machine takes 2 minutes. But the square is perfect,
this problem occurs in the range f540 to f600.
Outside these speeds, the car runs smoothly in speed.
I also checked with an oscilloscope, the frequency of the pulses on the pin out of arduino mega is half.
THANKS GREETINGS MATTEO.

78Matteo78, 2021-02-17:
I HAVE USED DIFFERENT SOFTWARE BUT IN ALL IT IS THE SAME (GRBLPLOTTER AND GRBLGRU)

FIRMWARE VERSION: grbl-Mega-5X-v1.1l.20190605.hex

GCODE:

( Simple Shape by GRBL-Plotter )
(<Header >)
( G-Code lines: 17 )
( Pen Down/Up : 1 times )
( Duration ca.: 1.0 min. )
( Conv. time  : 00:00:00.0299021 )
(</Header >)
F500 (Setup - GCode-Header)
G90
G00 Z2.000
M3 S1000 
G04 P1 
(<Figure Id="1" >)
G00 X0.000 Y0.000  
G01 Z-0.100 F200 
M3 S1000 
G01 X0.000 Y150.000 F600 (cw 1)
G01 X150.000 Y150.000  (cw 2)
G01 X150.000 Y0.000  (cw 3)
G01 X0.000 Y0.000  (cw 4)
M3 S1000 
G00 Z2.000 
(</Figure>)
M05
G0X0Y0 (Setup - GCode-Footer)
M30

fra589, 2021-02-18:
Hi @78Matteo78,  
  
I just reread the speed management code and I didn't see directly where your problem could come from.  
With the elements you provided, I will be able to try to reproduce it on my machine, but do not be in too much of a hurry. I am currently finishing a major update of my other cn5X++ software and I do not want to disperse myself for to be able to remain efficient.  
In the meantime, can you confirm that you are not using cloned axes?  
  
@++;  
Gauthier.

78Matteo78, 2021-02-18:
Grazie... Nessun problema... Vai con Calma

78Matteo78, 2021-02-18:
thanks, no problem, take your time

78Matteo78, 2021-02-18:
[image]
$I command

fra589, 2021-02-22:
Hi @78Matteo78,  
  
I just tested on my machine today, and all speeds have a correct rendering.  
I can't understand what happen to you...  

But your grbl-Mega-5X version is not the latest, perhaps an update will be benefit for you.
  
@++;  
Gauthier.

78Matteo78, 2021-02-22:
ok thanks I'll try

78Matteo78, 2021-02-23:
Hi
I tried with the firmware update but nothing has changed.
Could it be the fact that I use Arduino mega clone, not original, you have original ??
or i will try to shield the arduino board to see if it improves.
Or power it with external 5 volts .... if you have any suggestions ???
Thanks, good job hello

fra589, 2021-02-24:
Hi @78Matteo78,  
  
I use different Mega cards, original or clone, and I have never had this type of problem. I find it hard to imagine the possible causes of your problem.  
Maybe it could come from the program you are using to send the GCode?  
It is possible that, if this program interrogates the Arduino board too frequently, the CPU is no longer able to process all the interrupts necessary to manage, on the one hand, the motor pulses, and on the other, the serial communication.  
You should try with the simple streaming script provided as an example with Grbl which you will find here:  
https://github.com/fra589/grbl-Mega-5X/blob/edge/doc/script/simple_stream.py  
  
@++;  
Gauthier.

78Matteo78, 2021-02-24:
Hi
i used 3 different software: grbl plotter, grblgru, grbl controller app on android smartphone.
All 3 with the same problem, I compiled gcode from multiple software but the problem remains, I changed some setup parameters $ to no avail.
I'll see in the future, for now it's not a problem
Thanks again, bye

closed 2022-06-26 by fra589