fra589/grbl-Mega-5X #181: How to wake up Mega-5X after switching the power off/on

GrblGru, 2021-03-02:
Hi Gauthier,

can you please tell me how to wake up Mega-5X after switching the power off and on again ?
I am using your great program with my program GrblGru and the users always have to press the red reset button on the Mega at the 1st power on to get the connection.
After that, you can interrupt the software connection as often as you like and switch it on again without any problems. 
But once you disconnect the power supply or the USB cable, you have to press the reset button again. 
It would be very nice if I could send a command via the interface so that Mega-5X reports ready.

Thanks for your help. Best regards
GrblGru

fra589, 2021-03-02:
Hi @GrblGru,  
  
I'm not sure I understand your problem...
In cn5X ++, I systematically send 2 carriage returns each time the serial port is opened to ensure that Grbl wakes up.  
You can also try sending the soft reset (^ X) command, which will trigger Grbl to send the initialization string (Grbl 1.1q ['$' for help] for the last version of grbl-Mega- 5X).  
  
@++;  
Gauthier.

GrblGru, 2021-03-02:
Thanks for your quick reply.

Of course I have already tried to send different commands. But I have the impression that Mega-5X is not ready yet. 
After opening the SIO I send the following commands:
             Sio.Write("$X\n"); // Kill alarm lock
             Sio.Write("\x18"); // Soft reset
             Sio.Write("$X\n"); // Kill alarm lock
           
After that I wait for Mega-5X to respond with the version string.
On Arduino UNO with GRBL I treat the interface the same way and there it works.

I still use an old version V1.1.L and flashed the Mega with XLoader. 

Do you still have an idea ?

fra589, 2021-03-02:
Hi @GrblGru,  
  
It seems to me that GrblGru is running on Windows...  
I had a somewhat similar problem on Windows when using the QSerial object from QT5 to communicate in cn5X ++. But adding a sending of "\r\n\r\n" solved my problem, until QSerial completely stop working like I wanted at all as a result of a Windows update. I then changed and used pyserial instead, and I've never had a problem since.  
I always initialize the connection with:  
`self .__ comPort.write (("\ r \ n \ r \ n"). encode ('utf-8')) # Wakeup Grbl`
just after the "`self .__ comPort.open ()`" instruction.  
  
With Linux, I never had a problem receiving the init string from Grbl.  
  
What is your Windows version?
What language / framework do you use for the development of GrblGru?  
Do you use a particular object for serial communication?  
  
@++;  
Gauthier.

walterwissmann, 2021-03-03:
Hi to all,
I come from the practice...
On no time i have a problem with grbl 5x at the last years with broken communications.
Sorry about my broken English...
So i sended files generatet by Freecad Postprozessor grbl. 
I sended Files/gcode with bcnc...
No problems...
So when the communication was down, where are the problem to start anithing new ?
When you have to restart anythink, your communikation to grbl5x are wrong...

Bests
Walter

GrblGru, 2021-03-03:
Thanks for the tip. I made some more attempts, also with ("\r\n\r\n"), but unfortunately without success.

Yes, my program runs on Windows from XP to Win 10.
But there is the same problem everywhere. Also on different Mega boards. I also have the information from a user that he has the same problems. He has also tested different programs ( "GRBL control", "Candle" and "Laser Grbl"). With some there are the same problems as with GrblGru, with others it runs without problems.

I suspect that it is due to the way the SIO is opened. However, I don't understand why it runs error free on UNO with GRBL.
I only use C# and the .NET Framework 4 and the function "SerialPort". 
I will make further attempts.

Thanks for your help :)
GrblGru

fra589, 2021-03-03:
Hi @GrblGru,  
  
Can you try with the original gnea/grbl-Mega from https://github.com/gnea/grbl-Mega to see if it's the same behavior?  
  
@++;  
Gauthier.

GrblGru, 2021-03-03:
Good idea. Just tried the version once. It is the same behavior. The very first time after plugging in the USB cable it does not work. If I then press the red reset button once it will work for 1000 years. I can even quit my program and restart it. It works. But I am not allowed to disconnect the USB cable.

I am now sure that it is not your program. If you still want to try it once. I use Visual Studio Professional 2012.
This is the principle code:

            SerialPort Sio = new SerialPort();

            Sio.PortName = Port;
            Sio.BaudRate = BaudRate;
            Sio.Parity = Parity.None;
            Sio.StopBits = StopBits.One;
            Sio.DataBits = 8;
            Sio.Handshake = Handshake.None;
            Sio.ReadTimeout = 10000;

            Sio.Open();

            Sio.Write("\r\n\r\n");
            Thread.Sleep(200);
                

Thanks for your time :)
GrblGru

fra589, 2021-03-03:
I thought the original gnea / grbl-Mega version would have the same problem...
I never had problem with Linux, neither with C, C++ or Python.
I rather think that the problem comes from the side of Windows, the usb / serial driver or the SerialPort class + System.IO.Ports module of Visual Studio.
I will try to test this as soon as I find a little time... 
@++;
Gauthier.

closed 2022-06-26 by fra589