fra589/grbl-Mega-5X #182: Grbl Controller

Tjugt, 2021-03-10:
Hello everyone,

first of all thanks for all the work you put in. I am a newbie in this field and I am only now building my own cnc.

Now while triing to get my setup to work I got frustrated because I used "Grbl Controller" and it seems like the grbl-Mega-5x has communication issues with this controller (and therefore I thought the Arduino Mega wasn't flashed correctly). I am able to turn the motors and all but it is not able to synchronise the position the carriage is in (as an example). With other grbl setups I didn't have such problems.

Is the Grbl Controller just not suited for this grbl setup?
Which controller is best and easiest to use?

Thanks in advance.
Tjugt

fra589, 2021-03-13:
Hi @Tjugt,  
  
It seems Grbl Controller is no longer maintained and so, it doesn't appear as good option to use Grbl today.  
See the last original source of it here: https://github.com/zapmaker/GrblHoming/blob/master/README.  
  
There are many other GCode controlers, but I don't have good knowledges about any other Grbl sender than my own cn5X++: https://github.com/fra589/cn5X.  
I began the cn5X++ project in 2018 because when I updated the original Grbl to make the grbl-Mega-5X to use up to 5/6 axis on milling CNC machine, all the Grbl senders were limites to 3 axis.  
cn5X++ is still in developpement, but I aleready use it for real CNC milling.  
  
Since you say that you are a beginner, I can only advise you to search among all the documentation already existing on the Internet, and to read as much information as possible. Many people have already asked themselves the same questions that you are asking yourself and have already obtained answers that you just need to read.   
  
@++;  
Gauthier.

Tjugt, 2021-03-18:
Hi @fra589,

Thanks for your reply.
I'll have a look at your Grbl sender 👍 

Chears,
Tjugt

closed 2021-03-18 by Tjugt