fra589/grbl-Mega-5X #183: Machine Homes fine, and Y and Z work fine with G0 commands, but X gives alarm 2 no matter what.

dgruhl, 2021-03-14:
Good morning and thank you for a most excellent project! I have been trying to convert my older CNC to a 4 axis machine, and in doing so am moving over to grlb-Mega-5x on a RAMPS1.4 board. My configuration has a cloned X axis (1 and 4), Y is 2 and Z is 3. So far I have the machine homing just fine, and moving with G0 commands on the Y and Z axis work great! However, every time I try and move the X axis I get an "alarm 2". This is true for both positive and negative moves.

My configuration is

$0=10
$1=255
$2=0
$3=0
$4=0
$5=1
$6=0
$10=1
$11=0.010
$12=0.002
$13=1
$20=1
$21=0
$22=1
$23=3
$24=500.000
$25=8000.000
$26=250
$27=5.000
$30=1000
$31=0
$32=0
$100=80.000
$101=80.000
$102=2267.717
$103=80.000
$110=10000.000
$111=10000.000
$112=500.000
$113=10000.000
$120=800.000
$121=800.000
$122=300.000
$123=800.000
$130=450.000
$131=390.000
$132=85.000
$133=450.000

and my current $? is

<Alarm|MPos:0.0000,0.0000,0.0000,0.0000|FS:0.0,0|Pn:|WCO:0.0000,0.0000,0.0000,0.0000>
[HLP:$$ $# $G $I $N $x=val $Nx=line $J=line $SLP $C $X $H ~ ! ? ctrl-x]

Any thoughts where to look next on this would be very much appreciated!

             Thank you!
                Dan

dgruhl, 2021-03-15:
Note to past me from future me - when cloning axis it is really important that you invert the cloned axis (if needed) in BOTH $23 and $3. No clue why this would trigger a Alarm 2, but getting this right seems to have fixed that problem - now diving into the mysteries of the A axis :)

closed 2021-03-15 by dgruhl

fra589, 2021-03-15:
Hi @dgruhl,  
  
You are right! since the axis are cloned, they must be move together exactly by the same. So, they must have all their parameters the same.  
I'm glad you found it by yourself :-)  
  
@++;  
Gauthier.