fra589/grbl-Mega-5X #184: How to Interface step servo instead of stepper??

noton2, 2021-03-15:
@fra589 , I am planning for a high precession 5 axis cnc machine, so i hv chosen step servo motor over stepper motor. Servo drive setting will be position control mode.
My servo drive model - [EPS-B1 series](http://en.dorna.com.cn/a/EPS_B1/30.html) 
[Dorna EPS-B1 manual](https://eximautomation.de/mediafiles/Dorna/Anleitungen%20Englisch/EPS-B1%20Manual.pdf) .
We can control that with 2 way- 1. Pulse and dir signal
[image]
Or 2. serial connection using RS232.
[image]
 Now my questions are - 1. How to connect with arduino using 1st method??
2. What i hv to change in code for tx rx pin allocation and intrefacing , axis selection and addressing?

fra589, 2021-03-16:
Hi @noton2,  
  
I don't know how your step servos work, and I don't think it's possible to use the RS232 operating mode without rewriting most of the code.  
  
To operate the motors, Grbl uses 3 outputs per motor: "**Step**", "**Direction**" and "**Disable**".  
These outputs are connected to the driver which controls the stepping motors.  
"**Disable**" allows you to poweroff (or disable) the motor driver,  
"**Step**" send impulses to the driver. At each impulse, the driver must turn the motor by one step in the direction defined by the state of the "**Direction**" output.  
  
The pinout of the outputs is defined in the cpu_map.h file. It's explained in the wiki here: https://github.com/fra589/grbl-Mega-5X/wiki/grbl-Mega-5X-pinout  
  
The documentation that shows how to change it in cpu_map.h is here: https://github.com/fra589/grbl-Mega-5X/wiki/Pinout-mapping-in-cpu_map.h  
   
@++;  
Gauthier.

noton2, 2021-03-16:
@fra589 thanks,
step servo is nothing but closed loop motor with high rpm and positional accuracy.

closed 2021-09-19 by fra589