fra589/grbl-Mega-5X #188: 6th axis moves after cclicking connect on UGS

martinkombat, 2021-03-21:
I  have a 6 axes cnc machine that uses all 5 slots with an additional slot extender( connected to aux 3) on ramps 1.4 board .  Right after connecting grbl to UGS 6th axis motor starts working. The movement is inconsistent. Somestimes two full turns sometimes 1/10th of a turn but it does move. I cleared EEPROM which didnt change the sitution. At the moment as I stated before, as soon as I click on connect on UGS the 6th axis motor starts turning. I am lost for an answer to that and would appreciate even an opinion as to why this might happen. Thanks

fra589, 2021-03-23:
Hello @martinkombat,  
  
Since I programmed the sixth axis, I have never had the opportunity to fully test it. For this, I have to add the hardware to my test platform and I am running out of time for it.  
Considering your problem, I will have to do it :-)  
In the meantime, I need a little more details on your configuration in order to understand more easily.  
Please give me precisions by sending the output of the commands Grbl $ I and $$, and also if you have made any changes in the config.h and cpu_map.h files...  
  
@++;  
Gauthier.

fra589, 2021-04-29:
> Please give me precisions by sending the output of the commands Grbl $ I and $$, and also if you have made any changes in the config.h and cpu_map.h files...

No response? could it be that your problem has miraculously taken off?

closed 2021-04-29 by fra589