fra589/grbl-Mega-5X #190: G2/G3 Arcs with an XYUV Machine?

hexagon5un, 2021-04-05:
I have four-axis (2x2) hotwire cutter, and would like to know if it's possible to run arcs in both the XY and UV planes.

I've tried with like G2 X0 Y0 U0 V0 I10 F200, for instance, and the XY does a circle, but the UV stays still. 

Is this implemented, and if so, what's the right Gcode incantation?  If not, I can keep on approximating by straight lines.  :)

fra589, 2021-04-06:
Hi @hexagon5un,  
  
Unfortunately for you, it's not implemented...  
As you can read in the file gcode.c (line 919 to 1141), the arc calculation is made only in the XY, ZX or YZ planes...  
Planes mode 17.1 (UV), 18.1 (WU), and 19.1(VW) are not supported by Grbl.  
  
@++;  
Gauthier.

hexagon5un, 2021-04-06:
Thanks Gauthier!  

And thanks for the pointer to where it is in the code.  I may just have to have a hack at it and see what happens.

fra589, 2021-04-08:
Hi @hexagon5un,  
  
As I am retired, I unfortunately no longer have access to the standards defining the ISO GCode, and therefore, I cannot say which letters can be used to define the circular interpolation elements in UVW planes.  
In LinuxCNC which I use as a reference today, these features are not implemented:  
cf. http://linuxcnc.org/docs/2.8/html/gcode/g-code.html#gcode:g2-g3  
> 'Planes 17.1, 18.1, and 19.1 are not currently supported.'  
  
If you choose to use your own letters for these parameters (be careful, L and N are already used), you will also have to implement their use in the decoding part of the GCode sentence (between lines 302 and 403 of the gcode.c file)  
  
@++;  
Gauthier.

closed 2022-06-26 by fra589