fra589/grbl-Mega-5X #191: How to interface with a Scara robot - A, B and Z axis

tarsisprado, 2021-04-17:
I am developing a simple Laser Scara Robot (two rotating joints and a linear Z axis). I am new here and I am starting to use grbl now. As my robot only has three axis, I would like to use grbl-Mega with Arduino Uno and Uno CNC shield. Is it possible? The grbl for Arduino Uno can only control 3 linear axis until I know.
Coud you please help me interfacing with my Scara robot? Any help would be really appreciated.

Thanks in advance.

fra589, 2021-04-17:
Hi @tarsisprado,  
  
The Grbl-Mega version of Grbl work only on the chip Atmel AVR ATmega2560 which has more resources (FLASH, SRAM, and EEPROM) than the 328p.  
  
My Grbl-Mega-5X version work weel with the RAMPS 1.4, 1.5 or 1.6 shield which is not much more expensive than the Uno + CNC Shield.  
  
You can do what you whant by editing the config.h file with this:  
  
```
(...)  
  #define N_AXIS 3            // Number of axes
  #define N_AXIS_LINEAR 1     // Number of linears axis
(...)  
#define AXIS_1_NAME 'Z' // Axis names must be in X, Y, Z, A, B, C, U, V & W.
(...)  
#define AXIS_2_NAME 'A'
(...)  
#define AXIS_3_NAME 'B'
(...)  
```
  
The Z axis must be the firsh axis as linears axis must be located before the rotating axis in the internals calculation structures.

@++;  
Gauthier.

tarsisprado, 2021-04-17:
Hello Gauthier,

Thanks very much for the quick reply. I also have Mega and Ramps 1.4 here. I will try the changes you suggested in config.h, seens very easy! 
Thanks again for your support.

fra589, 2021-04-18:
You're welcome

closed 2021-04-18 by fra589