fra589/grbl-Mega-5X #195: Problem when changing axis names to E and H

korhanmd, 2021-04-29:
Hi,

There is a robotic arm in GrblGru and its axis names are A, B, C, D, E, H. I wanted to try it and changed axis names in `config.h` according to this. Changed `N_AXIS` to 6 and `N_AXIS_LINEAR` to 0. When I move the axes, it works perfectly from axis A to D, but if I send a command for E or H, it hangs and doesn’t respond until I reset the connection.

First, I suspect if the axis names are not valid. Because in `config.h` there are X, Y, Z, U, V, W, A, B, C letters as axis names. Then started to test D axis which is not listed in `config.h` too but it was always working properly.

I was using the software without problem while it has 5 axes and 3 of them are linear. I checked if it cause a problem when all axes are rotational. Changed `N_AXIS_LINEAR` from 0 to 6 gradually but nothing has changed. I changed order of the axes as D, E, H, A, B, C and got the same result.

Then, I thought maybe 6 axes are too much. I changed my `N_AXIS` to 3 and used D, E, H as axis names.  Nothing has changed again.

At last, I suspect that the parser has misinterpreted these commands. I added `serial_write` commands inside the `protocol_main_loop` function in `protocol.c` to see where it goes. It worked properly until the movement started. It goes inside the while loop, reads the letters, reads `‘\r’` at the end of the command, executes the line and even gets OK message back. It resets the tracking data for next line and completes the if statement inside while loop. But then all communication goes.

I also tried to send `G0 E0` and `G0 H0` commands when they are in the zero position. Parser interprets the commands, turns OK message and communication keeps working properly when it doesn’t need to move. It seems like the problem occurs when these axes try to move. Their step configurations are same as the other properly working axes.

Are these axis names unusable? If not, what could be the cause of this problem and how can it be solved? Thanks.

Best regards,
Korhan

RaphaelDives, 2021-04-29:
Hey @korhanmd,
just a little idea/thought, that you checked for shure... ;)
But did you check the $$ configuration for the additional axes? 
(I would also do an $RST after changing anything on axes.)
Cheers
Raphael

fra589, 2021-04-29:
Hi @korhanmd,  
  
The `N_AXIS_LINEAR` definition is only used when the mm to inch transform is needed. When the G20 unit mode is active, the conversion is only aplied on axis if their number is <= N_AXIS_LINEAR. So, it doesn't have any influence on your problem.  
  
Axis names other than X, Y, Z, A, B, C, U, V and W have not been tested, but pherhaps they can be work if their letter in not used by other gcode items...  
  
I think D and H should work, but I have a doubt about E which may be interpreted as scientific number notation when decoding the serial flow?  
  
I'm not at home, so, I can't test and verify...  
  
To verify the letter usability, you shoud try with  axis:  
first with X, Y and D  
then X,Y and E  
finally X, Y and H...  
  
Don't forget to reset Grbl's parameters with `$RST=*` each time you change the axis definition after flashing your board. Without this, Grbl may have unpredictable behavior.    
  
Tell me about the results of your next tests, I will check this when I will return to home next week.  
  
@++;  
Gauthier.

korhanmd, 2021-04-30:
Hi @RaphaelDives and @fra589,

Thank you for your responses. I checked `$$` and made lots of tweaks on parameters for each configuration to be sure.

I made 3-axis XYD, XYE, and XYH configurations. All worked without problem. It seems like these letters can be used as axis names. Then I tried 3-axis DEH configuration with `$RST=*` command and it didn't work again. To check if it needs X and Y axis to work properly I used 6-axis XYZDEH configuration. This configuration didn't work, too.

After these tests I suspected it doesn't allow E and H axis at the same time and tried 5-axis XYZDE and XYZDH configurations. They didn't work either. As there is no Z axis in the working configurations I tried 5-axis XYDEH and 4-axis XYDE and XYDH configurations but nothing changed.

Then I tried to remove D axis from configurations because in the working configurations D, E, and H axes were also not used together. I tested 5-axis XYZEH and 4-axis XYEH configuration. This time E axis worked but the problem continued on H axis. I tried 4-axis XYZH combination and it worked.

At last I tried to change order of E and H axes in configurations XYZEH and XYEH. I made them XYZHE and XYHE. This time H axis worked and E axis didn't. Apparently it only allows one letter different than XYZABCUVW as axis name. The one which comes first is working properly and others aren't.

 I hope this information helps you with your tests. I will continue to search the source and the solution of the problem, too.

Best regards,
Korhan

fra589, 2021-05-10:
Hi @korhanmd,  
  
I just pushed a new version (v1.1t) which add the ability yo use X, Y, Z, U, V, W, A, B, C, D, E & H as axis names.
The main problem was the axis names D, E and H was not known in the homing algorithm.
  
I made a test with this config.h definitions:
```
#define N_AXIS 6            // Number of axes (3 to 6)
#define N_AXIS_LINEAR 0     // Number of linears axis
(...)
#define AXIS_1_NAME 'A' // Axis names must be in X, Y, Z, A, B, C, U, V, W, D, E & H.
(...)
#define AXIS_2_NAME 'B'
(...)
#define AXIS_2_NAME 'C'
(...)
#define AXIS_2_NAME 'D'
(...)
#define AXIS_2_NAME 'E'
(...)
#define AXIS_2_NAME 'H'
```
But with only 4 reals motors because I don't have enought of it...  
  
The 4 motors are moving normaly, so I think it will be OK for 6.  
  
@++;  
Gauthier.

fra589, 2021-05-14:
Hi @korhanmd,  
  
Any clues? Have you fully tested the last version which implement adding D, E & H axis?  
  
@++;  
Gauthier.

korhanmd, 2021-05-14:
Hi @fra589,

Thank you for your help. I'm in quarantine right now and I can't try it with the motors, too. But I tried with only board and it seems like the communication works as it should after giving commands for these axes. I don't think there will be a problem with the motors.

Best regards,
Korhan

closed 2021-05-14 by korhanmd