fra589/grbl-Mega-5X #197: g-code locked out during alarm or jog state. error:9 lightburn

CityLotHomestead, 2021-05-13:
I just bought an Ortur Laser Master 2 20w and its like the the laser in the axis assembly is going straight to the right and hitting the end, then errors out. What do you believe it is?
Thank you for your time.
Amanda

ok 
Starting stream 
Layer C00 [SC7A20:(365) 
Shock and Movement detected!] 
Ortur Laser Master 2 Ready! 
OLF 140. 
Grbl 1.1h ['$' for help] 
[MSG:'$H'|'$X' to unlock] 
error:9 G-code locked out during alarm or jog state. 
On or near line 105: 
Job halted 
Stream completed in 0:03 
Starting stream 
Layer C00 
error:9 
G-code locked out during alarm or jog state. 
On or near line 0:

fra589, 2021-05-13:
Hi @CityLotHomestead,  
  
At Grbl startup, error 9 is normal: When powering up, Grbl enters Alarm mode to force the origin point because it cannot know its position.
To unlock, all you have to do is running the homing procedure (Grbl command `$H`). You can also unlock the alarm with the `$X` command, but be careful, this does not allow Grbl to know its position. So you will have to make sure by yourself that the Gcode program does not exceed the limits of your machine.  
It is also possible to deactivate the Alarm mode on power-up by deactivating the homing in the parameters of Grbl (`$22 = 0`).
It doesn't seem to me that the version of Grbl used on your Otur Master 2 machine is grbl-Mega-5X. I couldn't help you more, I don't know the other versions as well.  
  
@++;  
Gauthier.

CityLotHomestead, 2021-05-14:
Thank you so much! I truly appreciate you. It is a Laser Master 2 20w.

closed 2021-09-19 by fra589