fra589/grbl-Mega-5X #198: Dual laserhead control

martinjurekcz, 2021-06-01:
Hi, is there any way for grbl to control two lasers in one workinghead and the same time? What must happen for it to work? Can anyone help please? :)

fra589, 2021-06-01:
Hi @martinjurekcz,  
  
I understand than the 2 lasers are in the same head and move in the same movement, so, you only need X and Y for working and Z for adjust the laser height.  
So, your need is to have 2 PWM outputs for the 2 laser. In Grbl, the PWM is managed by the spindle_control.c. You will need to add your own code in it for adding the second laser.  
The laser is drived in Gcode by the spindles commands (M3, M4 & M5 and the S code for power). You also need to add your own GCodes in the gcode.c file to add the second lasez commands.  
  
@++;  
Gauthier.

martinjurekcz, 2021-06-02:
Thanks a lot. I will try it and let know how it worked. I think this could be useful for many other semiconductor CNC laser owners. It could easily double the productivity of our machines :)) That would be awesome since semiconductor lasers are really slow(small power output).

martinjurekcz, 2021-07-14:
Well, addling additional gcode commands and spindle control proved to be not that easy as I though. Maybe it is not necessary since there would be just one motion. Is there any way for direct pin control in grbl? Something like M42. Could you help with this question @fra589 please?

fra589, 2021-07-14:
Hi @martinjurekcz,  
  
This should not to be too difficult to add direct output pin control M commands in the gcode parser...  
  
As I try to have a maximum of LinuxCNC compatibility, I can add the M64/M65 codes.  
cf.: http://linuxcnc.org/docs/html/gcode/m-code.html#mcode:m62-m65.  
M62/M63 will be a bit more complicated, but not too hard :-)  
  
I can do it a little bit later as I'm overbooked at this time (I have to prepare for some aeromodelling international contests and for the french championships which will be on august from 3 to 11...  
I will can only do it after august. I hope you will can wait for it, perhaps on the begining of september...  
  
@++;  
Gauthier.

martinjurekcz, 2021-07-14:
Hi @fra589, that would be fantastic. Beginning of September sound perfect. Is it possible to implement M68 rather than M64/65 since it has the analog PWM functionality(is it big difference)? I believe that for homogeneous and linear motion(as raster engraving is) the immediate version could work nicely. Such a feature could help many projects of grbl on Arduino MEGA since there should be additional PWM pins free which could be used for lasers, fans, servos and many others. That's wonderful..

fra589, 2021-07-14:
Hi @martinjurekcz,  
  
M68 is much more complicated since PWM use interrupt which creates a performance and/or conflict risk. Interrupts take priority over the rest of the code, we need to verify if all other codes have time to go and if there is no conflict between interrupts... But I will try it :-)
  
Since I have a lot of other things in mind at the moment, I might forget about this subject by the end of the summer ...
Remind me at the beginning of September to come back to this topic :-) 
  
@++;  
Gauthier.

martinjurekcz, 2021-09-06:
Hi @fra589 How was the summer? How do you feel about M67/68 functionality for grbl? I was also thinking about buffer. I found an information that buffer is only for G command and not for M commands. Is it the same case for grbl? is it possible to buffer also M67/68? the motion would probably jitter without buffer.

fra589, 2021-09-13:
Hi @martinjurekcz,  
  
G and M commands are buffered in the same way in Grbl.  
I started writing the code for M62-M65, it is fully integrated with the GCode parser and optimized like any other command.
  
@++;
Gauthier.

SolveEtCoagula07, 2021-09-19:
Hello @fra589,
Out of curiosity, do you know when the M62/M63 feature will be published?

fra589, 2021-09-19:
Hi @martinjurekcz, hi @SolveEtCoagula07,  
  
I just published the new v1.2a version which implement the M62 to M65 digital output in grbl-Mega-5X.  
  
As described in http://linuxcnc.org/docs/html/gcode/m-code.html#mcode:m62-m65:  
- M62 P- - turn on digital output synchronized with motion. The P- word specifies the digital output number.
- M63 P- - turn off digital output synchronized with motion. The P- word specifies the digital output number.
- M64 P- - turn on digital output immediately. The P- word specifies the digital output number.
- M65 P- - turn off digital output immediately. The P- word specifies the digital output number.
  
I also updated the wiki page : https://github.com/fra589/grbl-Mega-5X/wiki/grbl-Mega-5X-pinout  

I hope this will work fine for you, test it and tell me if it work good.  
  
@++;  
Gauthier.

SolveEtCoagula07, 2021-09-25:
Hello @fra589,
I have tested the M62/M63 options and they seem to do exactly as I require!  On another note, is there a way to know when motion is complete?  Say I send G0X2000...once X reaches 2000, could the program send a "completed action" message?  This way I would know when I can send the next command.  Or does the buffer do this already?

Thank you for your amazing work on this repo!

fra589, 2021-09-26:
Hi @SolveEtCoagula07,
M62/M63 are synchronized with motion. This mean that if you send:
```
M62P0
G91G1X1000F100
M63P0
```
The result will be:
- Turn on the digital output 0,
- Move X by 100 at speed 100,
- Turn off digital output 0 at the end of the buffered X movement.

Conversely, if you need to turn on or off digital outputs immediately (without waiting the end of buffered motions), you can use M64/M65.

@++;
Gauthier.
P.S. Don't forget to close the issue when all is OK for you.

martinjurekcz, 2021-09-26:
hi @fra589, thanks a lot for new version with digital control. Im just coming back from holidays - im looking forward to implement it. Is there some way how analog output pins (PWMs) could be controlled with this procedure? In order to control servos, fans, lasers and other devices? Or this would be too difficult? Thanks a lot one more time..

fra589, 2021-09-27:
Hi @martinjurekcz,  

Assign PWM output is rather more complicated than digital outputs.
Implementation of [M67/M68](http://linuxcnc.org/docs/2.8/html/gcode/m-code.html#mcode:m67) is added to the TODO list, but without prevision date...

@++;
Gauthier

martinjurekcz, 2021-10-30:
Hi @fra589,
I understand, no problem. Is it something which could be expected within a year perhaps? Or do you think it's more distant future? :) thanks for any kind of forecast. 
nice rest of the weekend :)

fra589, 2021-11-04:
Hi @martinjurekcz,  
  
I probably work on this feature in the first quarter of 2022...  
So, many less than one year :smiley:   
  
@++;
Gauthier.

martinjurekcz, 2021-11-05:
Hi @fra589, 

That's amazing! Will wait for it. Thanks a lot.

evilhope, 2022-01-18:
Hi @fra589 @martinjurekcz how are you, I hope you are fine.
I am trying to use M64P0 command but terminal returns me an error 20 unsupported command.
Best wishes and wait your answers!

fra589, 2022-01-19:
Hi @evilhope,  
  
Do you use the latest version? Can you post the grbl `$I` result?  
The digital output have been implemented in the 1.2a version, and a mistake was introduced which give error 20 when using line numbering.  
This bug have been patched in the 1.2d version.  
  
@++;
Gauthier.

evilhope, 2022-01-20:
Hi @fra589
The result was 1.1g... version. Then when I cloned the repo the version changed to 1.2d and M62-M65 functionality worked. Thanks!

evilhope, 2022-04-06:
Hi @fra589 
How are you, I wish you are fine. I am here again. My cnc has **double spindle(double Z axis)**. I am using A **axis as a cloned Z axis** for second spindle. I have a question, is it possible to disable this cloned Z axis while use other Z axis?. I want to use only one spindle if I don't  want to cut with two spindles(duplicate cut). Thanks!

fra589, 2022-04-10:
Hi @evilhope,  
  
Axis cloning use compilation options which cannot be disabled other than recompile with other parameters.  
  
@++;  
Gauthier.

RaphaelDives, 2022-04-10:
Hey,
Just two thoughts:

You could add a switch for the enabled signal of the stepper driver. Then you can choose manually if one or two spindles should work. (Depending on your setup it could also be possible to reroute the enabled pin to an digital output and control it via software if you are fit enough to program it. Of course you can also use something different for that like cooling if you don’t need it…)

Second: Maybe it’s possible to work without cloning, with a second Z axis, and „clone“ it in your Gcode. But this is just an idea. I don’t know if it is possible to use two independent Z axis. 😉

Cheers
Raphael

evilhope, 2022-04-12:
Hi @fra589 and @RaphaelDives,
Thanks for your replies. @fra589 Thanks for your clarification
Awesome, I like your second idea!. I am using bCNC as sender. I think following your idea I can add a second Z axis into gcode as A axis. For example G0 X10 Y30 Z-5 A-5, I could use python to add this A gcodes where Z axis code exist. This way is correct or maybe I am screwing up because I finally don't full understand the code of grbl Mega 5X. Thanks for your time and sorry for my bad English.

fra589, 2022-05-16:
Hi @martinjurekcz,  
  
I just finished (and published few days ago) digital input/output new functionalities for grbl-Mega-5X, and now, I'm working on analogic input/output (M67/68 for output and M167/166 for input).  
I'm at the beginning of the process, I'm currently in re-reading the atmega2560 datasheet to verify the ability of the microcontroller to do it (I'm not sure that there is enough timer ressources to add another PWM which can work simultaneously with the all other implemented functions...  
I'm currently in writing some tests code to verify all of it...  
  
Results of those test will probably come soon.  
  
@++;  
Gauthier.

fra589, 2022-06-26:
Hi @martinjurekcz, @evilhope, @SolveEtCoagula07, @RaphaelDives,  
  
New M67 / M68 analog PWM output announcement here:  
https://github.com/fra589/grbl-Mega-5X/issues/112#issuecomment-1165787500

@++;  
Gauthier.

closed 2022-09-17 by fra589