fra589/grbl-Mega-5X #201: homing for A and B axis

naumaan9639, 2021-06-07:
hello, i am trying to make a 5 axis machine which have two rotary. when i press the homing button first z axis home and then X and Y home but A and B are not homing . how to make them home.

fra589, 2021-06-07:
Hi @naumaan9639,
  
Usually, rotary axis doesn't have limit switch, so by default, homing are not enabled for those axis in the config.h file.  
  
You need to edit config.h, you have 5 axis and you want to enable homing on A and B, so you need to uncomment the lines 181 and 182by suppressing the "//" at the begining of the line and give the correct number for the cycles. For your need, you should have:  
```  #define HOMING_CYCLE_0 (1&lt;&lt;AXIS_3) // Home Z axis first to clear workspace.
  #define HOMING_CYCLE_1 ((1&lt;&lt;AXIS_1)|(1&lt;&lt;AXIS_2))     // OPTIONAL: uncomment to move X,Y at the same time.
  //#define HOMING_CYCLE_1 (1&lt;&lt;AXIS_1) // Home X axis  // OPTIONAL: uncomment to move only X at a time.
  //#define HOMING_CYCLE_2 (1&lt;&lt;AXIS_2) // Home Y axis  // OPTIONAL: uncomment to move only Y at a time.
  #define HOMING_CYCLE_2 (1<<AXIS_4) // Home 4th axis (A)
  #define HOMING_CYCLE_3 (1<<AXIS_5) // Home 5th axis (B)
```
  
@++;  
Gauthier

naumaan9639, 2021-06-18:
Thanks @fra589

closed 2021-06-18 by fra589