fra589/grbl-Mega-5X #202: D6 PWM 50Hz

rcKeith, 2021-06-16:
Hi
Firstly many thanks for this firmware its made my CNC Foam cutter project possible and many have built it.  I'm starting a new project based on the CNC Needle Cutter for foam.  I'd like to be able to set the D6 spindle control to 50hz. I can see this in the cpu_map.h around line 264 there is a init mask. From the reading I've done, I think I need to apply a mask for the pre-scaler. But I'm not sure how I can get 50Hz to operate an RC servo.  I have connected a servo and it does move once but that's it. I think I need to use 5OHz if that's possible.
If you could help or point in the right direction I'd be very grateful
Keith

fra589, 2021-06-17:
Hi Keith,  

The spindle PWM configuration is made from line 273 to line 299 of cpu_map.h file. This configuration is used by the spindle_control.c file to drive the spindle output.
  
In Grbl, the spindle control use the "Fast PWM" mode. In this mode, the PWM frequency can only use some fixed values, depending of the prescaler defined by the SPINDLE_TCCRB_INIT_MASK value (cf. https://www.arduino.cc/en/Tutorial/SecretsOfArduinoPWM or https://forum.arduino.cc/t/mega-2560-pwm-frequency/71434/2 or many other Google's reply to the "arduino mega 2560 set pwm frequency" question).  
  
To change the PWM frequency and have an exact value of 50Hz, you must change the PWM mode to "phase-correct PWM" mode by changing its definitions in the cpu_map.h file and updating the spindle_control.c according your definitions changes.  
  
@++;  
Gauthier

rcKeith, 2021-06-18:
Thanks, I'll give it another try.
Keith

closed 2021-09-19 by fra589