fra589/grbl-Mega-5X #207: Additional IO pin

qiwalker, 2021-08-03:
Hi, 

how to get more output ?
currently i'm on progress to build custom PNP machine (Not for SMD), the machine is need 4 output control solenoid valve.

thanks
rizki

fra589, 2021-08-03:
Hi @qiwalker,  
  
As I said in the https://github.com/fra589/grbl-Mega-5X/issues/198 issue, I plan to add the functions M64/65 and M68 (see http://linuxcnc.org/docs/2.8/html/gcode/m-code.html#mcode:m62-m65).  
As I'm a little bit overbooked, I plan to do this after the begining of september...  
  
@++;  
Gauthier.

qiwalker, 2021-08-04:
hi @fra589 

thanks for your reply, okay i will wait for the next update

thanks ^_^

fra589, 2021-09-19:
Hi @qiwalker,  
  
As I mentioned in https://github.com/fra589/grbl-Mega-5X/issues/198#issuecomment-922484364, M62-M65 are added in the v1.2a version published today.  
  
@++;  
Gauthier.

closed 2021-09-19 by fra589