fra589/grbl-Mega-5X #208: Spindle pin from D8 to D11

d-portero, 2021-08-05:
Hello. I hate leaving an issue here, as I know there is somewhere out there that has this information. But I have been trying for weeks to switch my pin from D8 to D11 as I need 5V pwm control for my laser. 

I found issue #138 https://github.com/fra589/grbl-Mega-5X/issues/138 which is similar, but going to D6 instead. I would follow this and use D6, but my pin 6 seems to be broken. I have confirmed that my pin 11 is working, however.
I have tried to get all of the info from that issue, but I still seem to be missing something. 

I have changed my OCR register, as well as trying to change the TCCR register (which may be incorrect)

I would forever be in your debt I could get a little help with this. 

`  // Advanced Configuration Below You should not need to touch these variables
  // Set Timer up to use TIMER4B which is attached to Digital Pin 8 - Ramps 1.4 12v output with heat sink
  #define SPINDLE_PWM_MAX_VALUE     1024.0 // Translates to about 1.9 kHz PWM frequency at 1/8 prescaler
  #ifndef SPINDLE_PWM_MIN_VALUE
  #define SPINDLE_PWM_MIN_VALUE   1   // Must be greater than zero.
  #endif
  #define SPINDLE_PWM_OFF_VALUE     0
  #define SPINDLE_PWM_RANGE         (SPINDLE_PWM_MAX_VALUE-SPINDLE_PWM_MIN_VALUE)

  //Control Digital Pin 11
  #define SPINDLE_TCCRA_REGISTER    TCCR4A
  //#define SPINDLE_TCCRB_REGISTER    TCCR4B
  #define SPINDLE_TCCRB_REGISTER    TCCR1B //changed for D11
  //#define SPINDLE_OCR_REGISTER      OCR4C
  //#define SPINDLE_COMB_BIT          COM4C1
  #define SPINDLE_OCR_REGISTER      OCR1A //changed to controll D11 instead
  #define SPINDLE_COMB_BIT          COM4A1

  // 1/8 Prescaler, 16-bit Fast PWM mode
  #define SPINDLE_TCCRA_INIT_MASK ((1<<WGM40) | (1<<WGM41))
  #define SPINDLE_TCCRB_INIT_MASK ((1<<WGM42) | (1<<WGM43) | (1<<CS41))
  //#define SPINDLE_OCRA_REGISTER   OCR4A // 16-bit Fast PWM mode requires top reset value stored here.
  #define SPINDLE_OCRA_REGISTER   OCR1A // changed for pin D11
  #define SPINDLE_OCRA_TOP_VALUE  0x0400 // PWM counter reset value. Should be the same as PWM_MAX_VALUE in hex.

  // Define spindle output pins.
//  #define SPINDLE_PWM_DDR   DDRH
//  #define SPINDLE_PWM_PORT  PORTH
//  #define SPINDLE_PWM_BIT   5 // MEGA2560 Digital Pin 8
    #define SPINDLE_PWM_DDR   DDRB
    #define SPINDLE_PWM_PORT  PORTB
    #define SPINDLE_PWM_BIT   5 // MEGA2560 Digital Pin 11`

fra589, 2021-08-12:
Hi @d-portero,  
  
II am currently on vacation, I will get back to you as soon as possible.  
  
@++;  
Gauthier.

d-portero, 2021-09-19:
Hi Gauthier,

Hope you had a good vacation! 

Just want to check and see if there are any updates on this issue. 

Thanks!

fra589, 2021-09-19:
Hi @d-portero, 
  
Yes it is :-)  
  
I just commit the v1.2a version which implement the choice of running the spindle PWM output ON D8 (0 - 12 v on RAMPS, the default) or D6 (0 - 5v) or D11 (0 - 5v).  
The choice can be made in config.h, line 76 to 83.  
  
Tell me if it work fine for you.

@++;  
Gauthier.  
P.S. The main goal of this version was to implement the GCode [M62 to M65 digital output](http://linuxcnc.org/docs/html/gcode/m-code.html#mcode:m62-m65). You can also test it and tell me if it work fine.

d-portero, 2021-09-24:
Hey Gauthier,

I am pleased to report that indeed the spindle on D11 does work! Thanks for that

However, I have been struggling with getting motors to move with this current version. With my previous version of grbl-5x the motors work just as expected. With this version, they grind really bad or don't move at all. I have tried setting up the config files to match my old grbl files, and set them up as a brand new version and the issue still persists. I am at a loss and have been trying to find an answer since you updated the code. 

-Drew

fra589, 2021-09-24:
Hi Drew,  

It's strange since I don't changed anything about movement code...
What was your previous version ?
Is there a difference between your old settings and the new ones?
Compare the result of $$ before and after the update...

After changing the grbl version, it's advisable to reset the flash memory by issuing the Grbl command reset factory: `$RST=*`
Without this, Grbl may have an unpredictabe behavior.

@++;
Gauthier.

d-portero, 2021-09-25:
I had a random thought and that it was the only thing I changed is that I added a laser diode with the accompanying wires. I turned of pwm_on_D11 and my problems went away. Seems that it was interference from the PWM signal on the wires. 

Sorry for bugging you on this issue not related to your code. I appreciate you changing the code and updating it to allow for what I wanted. I appreciate the great work you have done!

duffel90, 2021-09-26:
D11  also generate problems with my system. So I uses D6 and it works fine. 
But if i power up my system the laser flashes up. So it seems the PWM D6 is default high. @fra589 Is it possible to set default D6 low?

fra589, 2021-09-27:
Hi @duffel90, Hi @d-portero,

OOPS Sorry! I wasn't paying attention when responding to @d-portero's request to move the Spindle pin from D6 to D11.
@duffel90, you opened my minds :-)
The timer used by the PWM output on D11 is the same as the one used to generate the motor pulses... So, we cannot use D11 to make a PWM output! 
=> I removed it from config.h.
stepper.c use timer0 and timer1, this forbid using PWM on D4, D11, D12 and D13. It remains the possibility on D2, D3, D5, D6, D7, D8, D9, D10.

@duffel90, PWM is default low, but the pin is assigned just before the call to the spindle_stop() function. So, it was possible than the ouptput was not 0 few microseconds between the assignation and the spindle_stop() call.
In last git push, I also forced PWM output to 0 before pin assignment to avoid flash on laser.
As I don't have a laser for testing, tell me if this work for you.

=> New version v1.2c pushed few minutes ago.

@++;
Gauthier.

fra589, 2021-09-28:
Hi Drew,

As you cansee in https://github.com/fra589/grbl-Mega-5X/wiki/grbl-Mega-5X-pinout, D5 is used for spindle direction signal. If you want to use D5 as spindle PWM, you first need to remap it to another location.
Changing the hardware mapping of Grbl in cpu_map.h is explained here: https://github.com/fra589/grbl-Mega-5X/wiki/Pinout-mapping-in-cpu_map.h
For example, if you want to remap the D5 spindle direction to AUX-1-D57 which is on the port F3, you need to change the lines 212 to 214 in cpu_map.h to:
```
  #define SPINDLE_DIRECTION_DDR   DDRF
  #define SPINDLE_DIRECTION_PORT  PORTF
  #define SPINDLE_DIRECTION_BIT   3 // MEGA2560 Digital Pin 57 - Ramps 1.4 AUX-1 Signal pin (A3/D57)
```
(Don't use D0 and D1 on AUX-1 as there are used by the serial com)

Then, you will can remap the spindle PWM to D5:
- D5 PWM use timer 3, D8 & D6 use timer 4. So, you need to replace 4 by 3 in all TCCR4*, OCR4* and COM4*,
- You also need to adjust the correct OCR register, for pin 5, it's OCR3A,
- As the port D5 is 8bits PWM, SPINDLE_OCRA_REGISTER need to use the ICR register, so, it will be ICR3,
- Finally, you will change the port definition in the same manner than all other ports in cpu_map.h.
The whole PWM on D5 definition will be:
```
    // PWM on D5:
    // Set Timer up to use TIMER3A which is attached to Digital Pin 5 - Ramps Servo 2
    #define SPINDLE_PWM_MAX_VALUE     255.0 // Translates to about 1.9 kHz PWM frequency at 1/8 prescaler
    #ifndef SPINDLE_PWM_MIN_VALUE
      #define SPINDLE_PWM_MIN_VALUE   1   // Must be greater than zero.
    #endif
    #define SPINDLE_PWM_OFF_VALUE     0
    #define SPINDLE_PWM_RANGE         (SPINDLE_PWM_MAX_VALUE-SPINDLE_PWM_MIN_VALUE)

    //Control Digital Pin 6 which is Servo 2 signal pin on Ramps 1.4 board
    #define SPINDLE_TCCRA_REGISTER    TCCR3A
    #define SPINDLE_TCCRB_REGISTER    TCCR3B
    #define SPINDLE_OCR_REGISTER      OCR3A
    #define SPINDLE_COMB_BIT          COM3A1

    // 1/8 Prescaler, 8-bit Fast PWM mode
    #define SPINDLE_TCCRA_INIT_MASK (1<<WGM41)
    #define SPINDLE_TCCRB_INIT_MASK ((1<<WGM42) | (1<<WGM43) | (1<<CS41))
    #define SPINDLE_OCRA_REGISTER   ICR3 // 8-bit Fast PWM mode requires top reset value stored here.
    #define SPINDLE_OCRA_TOP_VALUE  0xFF // PWM counter reset value. Should be the same as PWM_MAX_VALUE in hex.

    // Define spindle output pins.
    #define SPINDLE_PWM_DDR   DDRE
    #define SPINDLE_PWM_PORT  PORTE
    #define SPINDLE_PWM_BIT   3 // MEGA2560 Digital Pin 5
```

I haven't tested it... Hoping I am not mistaken,

@++;
Gauthier.

duffel90, 2021-10-04:
@fra589 
Version 1.2c works perfect
So after a few tests I found out that my laser driver board has a problem with the GND voltage. When the Arduino starts the laser flashes and even when the PWM signal was gone the laser was flashing (PWM Voltage 0,09 V). So I put a resistor between PWM and GND and the blinking stops at startup and without PWM wiring.

fra589, 2021-10-04:
Hi @duffel90,  
  
I'm glad it work good for you. 

Warning ! The 1.2C=c version have a bug (editing error of me) which forbid the use of numbering GCode lines...
Nxxx CGode sentence generates an error...

This is corrected in the v1.2d pushed 5 days ago.

@++;
Gauthier.

d-portero, 2021-10-07:
Unfortunately, the code to change to D5 didnt work. I have tried to verify the code against other forums and can't see where there is an error. Any chance something is missing? 

-Drew

closed 2022-06-26 by fra589