fra589/grbl-Mega-5X #209: Servo

KONO70, 2021-08-23:
Hello
Is it possible to control a servo motor?
I have a 5 axis robotic arm which I use with your version of GRBL which works awake

For the clamp I would have to control the servomotor

Thank you

fra589, 2021-08-24:
Hi @KONO70,  
  
It should be possible to add servo control to grbl-Mege-5X, but how busy I am at this time, it's not in my soon TODO list...  
  
@++;  
Gauthier.

KONO70, 2021-08-25:
Ok merci 
il faut modifier spindle_controle.c ? 
J'ai regarder avec grbl servo mais bcp trop de changement entre les 2 versions

alexc57, 2022-11-14:
avez vous trouvé la solution, j'ai une mks base 1.6, et impossible de piloter se fi...u servo?merci

fra589, 2022-11-21:
Bonjour @alexc57,  
  
J'ai introduit depuis la version 1.2g.20220620 la possibilité d'utiliser une sortie supplémentaire PWM (commandes GCode M67/M68). Fonctionnalité activable avec la directive `#define USE_OUTPUT_PWM` dans config.h.  
  
L'activation et le choix de la pin de sortie se fait dans config.h, les paramètres PWM sont dans cpu_map.h et le code est dans analog_control.c.  
  
En ajustant les paramètres PWM dans cpu_map.h pour régler (entre autre) la fréquence compatible avec le servo, il devrait être possible de le piloter.  
https://www.google.com/search?q=atmega2560+pwm+frequency  
https://www.google.com/search?q=atmega2560+pwm+servo  
  
@++;  
Gauthier.

alexc57, 2022-11-25:
hello,
merci pour les infos. j'ai justement tout essayé mais rien ne fonctionne.
je vais continuer à chercher.
c'est malheureusement plus compliqué que je pensais.

merci
alex

fra589, 2022-11-26:
Adding new enhancement to the TODO list to be able to drive servo with the M67/M68 PWM output function...

alexc57, 2022-11-27:
hello, j'arrive a faire varier maintenant la pin d11, de 0 a 5v comme pour un spindle ! par contre le servo ne bouge pas! une idee?