fra589/grbl-Mega-5X #210: ENABLE_SAFETY_DOOR_INPUT_PIN question

mzeddd, 2021-09-07:
Basically it is the same question as it was posted in parent repository

https://github.com/gnea/grbl-Mega/issues/143

Can somebody fix this probem?

fra589, 2021-09-13:
Hi @mzeddd,  
  
The ENABLE_SAFETY_DOOR_INPUT_PIN definition is not longer used in the code since the safety door is always active if it exist...  
I will have to think about deleting it in config.h during the next updates :-)  
  
The safety door pin is defined in cpu_map.h as the Arduino Pin A12 - RAMPS Aux 2 Port.
This is used in the system.c code, function system_control_get_state().
  
As it's defined on the hardware A12 pin, if you have the "MSG:Check Door" when you connect your RAMPS board, perhaps there is something connected to this pin on your RAMPS or this pin is short circuited on your RAMPS...

@++;
Gauthier

mzeddd, 2021-09-14:
Thanks for the answer.

Is A12 pin normally closed or opened?
On my Ramps 1.6 Plus this pin is not connected to any other pin. It is clear from both sides.

Should I connect it to any other pin (which one?) to get my Door problem solved or it means that I have faulty board?

Thanks,
Valery

fra589, 2021-09-14:
Hi @mzeddd,  
  
The safety door is normally open switch. It's seems that the fault come from your RAMPS board...
To solve your problem, you can change the pin assignation of the safety door input pin by editing the cpu_map.h file.
You need to stay on the Arduino's port K as all controls pins must stay on this port... So, you can chose the A13 port which in unused by Grbl.
For this, change the cpu_map.h file line 232 to :  
`#define CONTROL_SAFETY_DOOR_BIT   5  // Pin A13 - RAMPS T0 Port`
See https://github.com/fra589/grbl-Mega-5X/wiki/Pinout-mapping-in-cpu_map.h for more info about remapping input/output in Grbl.
  
@++;
Gauthier.

closed 2021-09-19 by fra589