fra589/grbl-Mega-5X #214: Wrong state RUN when jogging right after probing

joecooletti, 2021-09-22:
I think GRBL is reporting a wrong state when jogging right after a g43.2 probing cycle. The first jog command after G43.2 reports state RUN instead of JOG. Issuing a jog cancel in that first jog results in 'error:8'.

See attached log, line 21...

[grbl.log](https://github.com/fra589/grbl-Mega-5X/files/7213492/grbl.log)

joecooletti, 2021-09-23:
I dived into the code and found out that this behaviour is caused by the kind of motion stop when the probe switch is triggered. in _probe.c_ the flag EXEC_MOTION_CANCEL is set which causes the motion to stop immediatly. A following jog command is first correctly marked with state STATE_JOG in _jog.c_, but this state is instantly overwritten in _protocol.c_ (line 366) with STATE_CYCLE, which sort of executes a cylce start after a motion cancel.

So far, so bad. Unfortunately my C programming skills don't allow me to think about a fix of this issue :(

As a workaround my gui can command a jog without move ($J=G90X0F1)

fra589, 2021-09-23:
Hi @joecooletti,  

Thanks for reporting and searching about it...
I'm quite busy for now,  I will look for this issue as soon as possible.

@++;
Gauthier.

fra589, 2021-09-25:
Hi @joecooletti,

This is clearly a bug...
I tested the original [gnea/grbl-Mega](https://github.com/gnea/grbl-Mega) and he already have the bug :-(

I'm searching for it, but this piece of code is very complicated.

As you see, the system status is set to Run on line 366 of protoclo.c, but this append because steppec.c have set the EXEC_CYCLE_STOP system flag before !
So, with EXEC_CYCLE_STOP, protocol.c set the state to Idle. Then, next loop, it set the state to Run because the planer is not empty.
I don't know why steper.c set the EXEC_CYCLE_STOP system flag... I continue to search for it.

@++;
Gauthier.

fra589, 2021-09-26:
Hi @joecooletti,

I finally found the problem ! :smiley: 
The sys.step_control was not restored after the end of the probe cycle in motion_control.c

This is corrected in the new v1.2b grbl-Mega-5X version I pushed few minutes ago.

@++;
Gauthier.

joecooletti, 2021-09-27:
Hi Gauthier,

I did a short test, and it works now 👍 
Great, thx a lot

closed 2021-09-27 by joecooletti