fra589/grbl-Mega-5X #225: Machine locked during program execution

atomino-max, 2021-11-01:
I have a problem on my cnc, I have installed the last version , hoping to fix the problem, but also with older version the program is stopped and don't proceed.
At this point I think the problem is generated from the software that translate the program to gcode, Aspire 8.0, but I cannot understand why the grbl is locking...
Below the code catched from the terminal of bCNC together with the code remained in the buffer.
I hope somebody can help me to fix the problem.
Thanks in advance

Grbl 1.2d ['$' for help]

[MSG:'$H'|'$X' to unlock]
Grbl 1.2d ['$' for help]
[MSG:'$H'|'$X' to unlock]
ok

ok

[MSG:Caution: Unlocked]
ok
$#
[G54:-439.000,-439.000,-319.000]
[G55:0.000,0.000,0.000]
[G56:0.000,0.000,0.000]
[G57:0.000,0.000,0.000]
[G58:0.000,0.000,0.000]
[G59:0.000,0.000,0.000]
[G28:0.000,0.000,0.000]
[G30:0.000,0.000,0.000]
[G92:0.000,0.000,0.000]
[TLO:0.000]
[PRB:0.000,0.000,0.000:0]
ok
$G
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
ok
$H
ok
b'$G\n'
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
ok
$#
[G54:-439.000,-439.000,-319.000]
[G55:0.000,0.000,0.000]
[G56:0.000,0.000,0.000]
[G57:0.000,0.000,0.000]
[G58:0.000,0.000,0.000]
[G59:0.000,0.000,0.000]
[G28:0.000,0.000,0.000]
[G30:0.000,0.000,0.000]
[G92:0.000,0.000,0.000]
[TLO:0.000]
[PRB:0.000,0.000,0.000:0]
ok
b'$G\n'
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
ok
$G
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
ok
$H
ok
b'$G\n'
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
ok
G1X50Z280F800
ok
G92Z20
ok
$#
[G54:-439.000,-439.000,-319.000]
[G55:0.000,0.000,0.000]
[G56:0.000,0.000,0.000]
[G57:0.000,0.000,0.000]
[G58:0.000,0.000,0.000]
[G59:0.000,0.000,0.000]
[G28:0.000,0.000,0.000]
[G30:0.000,0.000,0.000]
[G92:0.000,0.000,260.000]
[TLO:0.000]
[PRB:0.000,0.000,0.000:0]
ok
b'$G\n'
[GC:G1 G54 G17 G21 G90 G94 M5 M9 T0 F800 S0]
ok
$G
[GC:G1 G54 G17 G21 G90 G94 M5 M9 T0 F800 S0]
ok
$$
$0=10
$1=255
$2=0
$3=3
$4=0
$5=1
$6=0
$10=19
$11=0.010
$12=0.002
$13=0
$20=1
$21=1
$22=1
$23=7
$24=100.000
$25=300.000
$26=250
$27=1.000
$30=255
$31=0
$32=0
$100=3840.000
$101=3840.000
$102=3840.000
$110=800.000
$111=800.000
$112=800.000
$120=5000.000
$121=5000.000
$122=5000.000
$130=440.000
$131=440.000
$132=320.000
ok
b'$G\n'
[GC:G1 G54 G17 G21 G90 G94 M5 M9 T0 F800 S0]
ok
G90
ok
T1
ok
G17
ok
G21
ok
G90
ok
G0Z20
ok
G0X0Y0S24000M3
ok
G0X22.8Y31.915Z10
ok
G0Z6
ok
G1Z-1.667F300
ok
G1X416.2F600
ok
G1Y50.915
ok
G1X22.8
ok
G1Y69.915
ok
G1X416.2
ok
G1Y88.915
ok
G1X22.8
ok
G1Y107.915
ok
G1X416.2
ok
G1Y126.915
ok
G1X22.8
ok
G1Y145.915
ok
G1X416.2
ok
G1Y164.915
ok
G1X22.8
ok
G1Y183.915
ok
G1X416.2
ok
G1Y202.915
ok
G1X22.8
ok
G1Y221.915
ok
G1X416.2
ok
G1Y240.915
ok
G1X22.8
ok
G1Y259.915
ok
G1X416.2
ok
G1Y278.915
ok
G1X22.8
ok
G1Y297.915
ok
G1X416.2
ok
G1Y316.915
ok
G1X22.8
ok
G1Y335.915
ok
<Run|MPos:-424.280,-418.395,-45.457|Bf:0,142|FS:1015,255>
<Run|MPos:-423.695,-417.576,-45.713|Bf:0,142|FS:1015,255>
<Run|MPos:-423.108,-416.755,-45.970|Bf:0,142|FS:1015,255>
<Run|MPos:-422.489,-415.888,-46.242|Bf:0,142|FS:1015,255>
G1X416.2
ok
G1Y354.915
ok
G1X22.8
ok
'utf-8' codec can't decode byte 0xb8 in position 37: invalid start byte
Esecuzione finita
2021-11-01 12:47:03.033135
Current: 170 [170]  Completed: 26% [16s Tot: 16s ]

--------------------------------------------------------------------------
commands remained in the buffer
--------------------------------------------------------------------------

G1Y373.915
G1X416.2
G1Y392.915
G1X22.8
G1Y411.915
G1X416.2
G0Z10
G0X19Y420
G0Z6
G1Z-1.667F300
G1X420F600
G1Y19
G1X19
G1Y420

fra589, 2021-11-01:
Hi @atomino-max, 

Can you post the wole GCode file ? or a link to it if it's too big...
  
@++;
Gauthier.

atomino-max, 2021-11-01:
Hi Gauthier,
this is the link :
https://www.mediafire.com/file/mket6n619yzv56x/Piano439_3d.gcode/file

fra589, 2021-11-04:
Hi @atomino-max,  
  
I checked your GCode file and it is correct, without any default.
I think the problem may come from a serial communication error that bCNC can't override...

Perhaps you can try to lower the serial speed (value is `#define BAUD_RATE 115200` in config.h)
Or try another gcode sender...

I used my own sender (https://github.com/fra589/cn5X) with Grbl in Check mode (`$C`) and your file worked fine!

@++;
Gauthier.

atomino-max, 2021-11-07:
Hi Gauthier,
I have tried to reduce the speed to 57600 then 19200 with bCnc and Ugs without fixing the problem.
I have installed your sender cn5X, at 115200 loading the same file but the system got stuck, see 
attached screenshot

[image]
[image]

.

fra589, 2021-11-07:
Hi @atomino-max,  
  
This is very strange.?.?.?.
Can you also send me a copy of the "Grbl communication" tab of cn5X?  
Then the the result of $i command...  
  
@++;  
Gauthier.

atomino-max, 2021-11-08:
To send $I I have closed the comunication

because the interface cannot accept any command.

$I

[VER:1.2d.20210930:]

[AXS:3:XYZ]

[OPT:VNMGH,35,255,120]

ok

grbl comunication tab

Grbl 1.2d ['$' for help]

[MSG:'$H'|'$X' to unlock]

[VER:1.2d.20210930:]

[AXS:3:XYZ]

[OPT:VNMGH,35,255,120]

ok

$#

ok

$$

$0=10 >> Step pulse time (microseconds) : Sets time length per step (Minimum 3usec).

$1=255 >> Step idle delay (milliseconds) : Sets a short hold delay when stopping to let dynamics settle before disabling steppers. Value 255 keeps motors enabled with no delay.

$2=0 >> Step pulse invert (mask) : Inverts the step signal. Set axis bit to invert (00000ZYX).

$3=3 >> Step direction invert (mask) : Inverts the direction signal. Set axis bit to invert (00000ZYX).

$4=0 >> Invert step enable pin (boolean) : Inverts the stepper driver enable pin signal.

$5=1 >> Invert limit pins (boolean) : Inverts the all of the limit input pins.

$6=0 >> Invert probe pin (boolean) : Inverts the probe input pin signal.

$10=254 >> Status report options (mask) : Alters data included in status reports.

$11=0.010 >> Junction deviation (millimeters) : Sets how fast Grbl travels through consecutive motions. Lower value slows it down.

$12=0.002 >> Arc tolerance (millimeters) : Sets the G2 and G3 arc tracing accuracy based on radial error. Beware: A very small value may effect performance.

$13=0 >> Report in inches (boolean) : Enables inch units when returning any position and rate value that is not a settings value.

$20=1 >> Soft limits enable (boolean) : Enables soft limits checks within machine travel and sets alarm when exceeded. Requires homing.

$21=1 >> Hard limits enable (boolean) : Enables hard limits. Immediately halts motion and throws an alarm when switch is triggered.

$22=1 >> Homing cycle enable (boolean) : Enables homing cycle. Requires limit switches on all axes.

$23=3 >> Homing direction invert (mask) : Homing searches for a switch in the positive direction. Set axis bit (00000ZYX) to search in negative direction.

$24=100.000 >> Homing locate feed rate (units (millimeters or degres)/min) : Feed rate to slowly engage limit switch to determine its location accurately.

$25=300.000 >> Homing search seek rate (units (millimeters or degres)/min) : Seek rate to quickly find the limit switch before the slower locating phase.

$26=250 >> Homing switch debounce delay (milliseconds) : Sets a short delay between phases of homing cycle to let a switch debounce.

$27=1.000 >> Homing switch pull-off distance (millimeters) : Retract distance after triggering switch to disengage it. Homing will fail if switch isn't cleared.

$30=255 >> Maximum spindle speed (RPM) : Maximum spindle speed. Sets PWM to 100% duty cycle.

$31=0 >> Minimum spindle speed (RPM) : Minimum spindle speed. Sets PWM to 0.4% or lowest duty cycle.

$32=0 >> Laser-mode enable (boolean) : Enables laser mode. Consecutive G1/2/3 commands will not halt when spindle speed is changed.

$100=3840.000 >> 1st axis travel resolution (step/unit) : 1st axis travel resolution in steps per unit (millimeter or degre).

$101=3840.000 >> 2nd axis travel resolution (step/unit) : 2nd axis travel resolution in steps per unit (millimeter or degre).

$102=3840.000 >> 3rd axis travel resolution (step/unit) : 3rd axis travel resolution in steps per unit (millimeter or degre).

$110=800.000 >> 1st axis maximum rate (unit/min) : 1st axis maximum rate. Used as G0 rapid rate.

$111=800.000 >> 2nd axis maximum rate (unit/min) : 2nd axis maximum rate. Used as G0 rapid rate.

$112=800.000 >> 3rd axis maximum rate (unit/min) : 3rd axis maximum rate. Used as G0 rapid rate.

$120=5000.000 >> 1st axis acceleration (unit/sec^2) : 1st axis acceleration. Used for motion planning to not exceed motor torque and lose steps.

$121=5000.000 >> 2nd axis acceleration (unit/sec^2) : 2nd axis acceleration. Used for motion planning to not exceed motor torque and lose steps.

$122=5000.000 >> 3rd axis acceleration (unit/sec^2) : 3rd axis acceleration. Used for motion planning to not exceed motor torque and lose steps.

$130=440.000 >> 1st axis maximum travel (unit (millimeters or degres)) : Maximum 1st axis travel distance from homing switch. Determines valid machine space for soft-limits and homing search distances.

$131=440.000 >> 2nd axis maximum travel (unit (millimeters or degres)) : Maximum 2nd axis travel distance from homing switch. Determines valid machine space for soft-limits and homing search distances.

$132=326.000 >> 3rd axis maximum travel (unit (millimeters or degres)) : Maximum 3rd axis travel distance from homing switch. Determines valid machine space for soft-limits and homing search distances.

ok

$H

ok

G1Z-50

error:22: Undefined feed rate,

Feed rate has not yet been set or is undefined.

G1Z-50F800

ok

T1

ok

G17

ok

G21

ok

G90

ok

G0Z20.000

ok

G0X0.000Y0.000S24000M3

ok

G0X22.800Y31.915Z10.000

ok

G0Z6.000

ok

G1Z-1.667F300.0

ok

ok

G1X416.200F600.0

ok

ok

G1Y50.915

ok

ok

G1X22.800

ok

ok

G1Y69.915

ok

ok

G1X416.200

ok

ok

G1Y88.915

ok

ok

G1X22.800

<Run|WPos:10.600,14.837,15.351|

ok

G1Y107.915

[GC:G1 G54 G17 G21 G90 G94 M3 M9 T0|WPos:9.255,12.956,15.940|Bf:28,255|FS:1015,255>

ok

ok

ok

G1X416.200

ok

G1Y126.915

ok

G1X22.800

<Run|WPos:14.200,19.874000]

ok

ok

ok

G1Y145.915

ok

ok

G1X416.200

ok

ok

G1Y164.915

ok

ok

G1X22.800

ok

G1Y183.915

ok

G1X416.200

ok

G1Y202.915

ok

G1X22.800

ok

ok

G1Y221.915

ok

G1X416.200

ok

ok

G1Y240.915

ok

G1X22.800

ok

ok

G1Y259.915

ok

G1X416.200

ok

G1Y278.915

ok

G1X22.800

ok

G1Y297.915

ok

G1X416.200

ok

G1Y316.915

ok

G1X22.800

ok

G1Y335.915

ok

G1X416.200

ok

G1Y354.915

ok

G1X22.800

ok

G1Y373.915

fra589, 2021-11-10:
hi @atomino-max  
  
It look like it's due to serial/usb communication errors between the computer and the Arduino board...
As you are on Linux, perhaps you can see on the dmesg Linux command output to see if there is a communication hardware problem?

@++;
Gauthier.

atomino-max, 2021-11-10:
Not easy to understand, no error in dmesg
I generate gcode with Flatcam, without problems.
I have generate with bCnc the same program, and it's working prpperly.
Thanks for your support I will try to workaround.

[image]
[image]

closed 2022-06-26 by fra589