fra589/grbl-Mega-5X #229: Resuming after a door/alarm state resets M7 (Coolant) state

amoineau, 2021-11-15:
Hello,

## Context

We are developing a machine that can cut and mark textile.
In our configuration, we are using M7/M9 to command a relay thanks to the 'coolant' output. This allows us to direct the spindle signal to the primary or secondary tool, and thus change tool easily from the g-code.

## Problem
Unfortunately we noticed that when a door sensor or our emergency button was triggered, GRBL would reset the 'coolant' state. In our case, it means that we revert to the default tool. So if we trigger a door signal mid-shape, we then resume with the wrong tool.

## What we tried
In **protocol.c**, We commented the de-energizing of the coolant when going into sleep state (line 622). So now the coolant isn't reset when triggering a door, however, when sending the _resume_ signal it resets a second time.

## What we need
We couldn't find a way to prevent this behaviour, could someone help us find a solution please.
We would like the coolant state to be unaltered by door/alarm/hold or any such signal.

Thank you in advance for your help.

fra589, 2021-11-15:
Hi @amoineau,  
  
The 1.2d version of Grbl-Mega-5X have the ability to drive up to 4 digital output with the GCode commands M62, M63, M64, M65 (see http://linuxcnc.org/docs/html/gcode/m-code.html#mcode:m62-m65).
The output are connected to the D16, D17, D23 and D25 pins (see https://github.com/fra589/grbl-Mega-5X/wiki/grbl-Mega-5X-pinout)
You can use those digital output to drive  your tool selection relay.
The digital output status is not altered by the door/alarm state.
  
@++;
Gauthier.

amoineau, 2021-11-16:
Hello,
Thank you for your quick response. I wish we started with that, unfortunately the machine and the board are made, wired... and shipped (only one copy fortunately). Classic "it's not ready/ we need to deliver" situation.
So even if it's a perfectly fine solution, I'm afraid we can't implement it unless it's really our last resort.

fra589, 2021-11-16:
Hi @amoineau,  
  
I think it's not a problem if you don't need a real coolant functionality...
Remapping pins is easy, you just have to change few lines in the cpu_map.h file:
https://github.com/fra589/grbl-Mega-5X/wiki/Pinout-mapping-in-cpu_map.h
So, you can remap the coolant output to any other not used pin, then remap one digital output pin to the already wired coolant pin.  
  
@++;  
Gauthier.

amoineau, 2021-11-16:
Yes I forgot that ! (most of our pins are actually remapped for the home made board)
The only downside I still see, is that we are using GRBL plotter and there are leds for the spindle and the mist which allow us to clearly see which tool is running. Since one of the tool is a laser (mostly invisible) it is very useful, if not mandatory, to have this kind of information on screen to quickly see if there is an error.

amoineau, 2021-11-17:
Ok, thank you for all these information ! I'll try to see what we can do.

svenhb, 2021-11-18:
Hi, I made the GRBL-Plotter and try to proccess your D0 to D3 pin states. But with your latest code (ver 1.2d) I get no accessory info at all. Also I can't find a associated #define in all the *.h files...
```
< Grbl 1.2d ['$' for help]
<Idle|WPos:-50.000,-45.000,-20.000,0.000,0.000|Bf:35,255|FS:0,0>
<Idle|WPos:-50.000,-45.000,-20.000,0.000,0.000|Bf:35,255|FS:0,0>
<Idle|WPos:-50.000,-45.000,-20.000,0.000,0.000|Bf:35,255|FS:0,0|WCO:50.000,45.000,20.000,0.000,0.000>
<Idle|WPos:-50.000,-45.000,-20.000,0.000,0.000|Bf:35,255|FS:0,0|Ov:100,100,100>
<Idle|WPos:-50.000,-45.000,-20.000,0.000,0.000|Bf:35,255|FS:0,0>
<Idle|WPos:-50.000,-45.000,-20.000,0.000,0.000|Bf:35,255|FS:0,0>

```

svenhb, 2021-11-18:
Ok I figured out, I have to switch the pin-states to get an accessory info, but following behavior is strange:
I can set D0 via M64 P0, but not clear via M65 P0, right after setting. 
I need to set another Pin e.g. D1 to get an update for D0 also.
After a reset there is no update of the states, will they be cleared?

fra589, 2021-11-19:
Hi @amoineau,  
  
The accessory state (A:...) is always sended by Grbl with the Ov: sentence.  
If you don't have accessory state in the same message than the Ov:, this mean that all the accessories are off.  
The Ov: is pushed after all accessories change.

```
M64P0
ok
?
<Idle|WPos:0.000,0.000,0.000,0.000,0.000|Bf:35,255|FS:0,0|Ov:100,100,100|A:D0001>
M65P0
ok5
?
<Idle|WPos:0.000,0.000,0.000,0.000,0.000|Bf:35,255|FS:0,0|Ov:100,100,100>
```

- The first '?' answer reflects the active status of digital output 0,  
- The second '?' reply mean than there is no accessory active, so, the digital 0 is off.  
  
@++;
Gauthier.

svenhb, 2021-11-19:
>If you don't have accessory state in the same message than the Ov:, this mean that all the accessories are off.

Yes I figured out later, also D0000 will never be sent...
After realizing this, I deleted my posts...

fra589, 2021-11-19:
Yes, it work like any other accessories states. 😃

amoineau, 2021-11-19:
Ok, I was kinda lost on what was happening, I received the mails but couldn't see anything here ^^

svenhb, 2021-11-19:
Yes, sorry, I deleted my posts

svenhb, 2021-11-19:
Thanks, now I can display the D0 - D3 status: https://github.com/svenhb/GRBL-Plotter/releases
Are there any other features (compared to standard grbl) that I am not aware of?

fra589, 2021-11-20:
Hi @svenhb,  
  
Grbl-Mega-5X have capability of 3 to 6 axis, capacity to rename axis (change letter of axis) and capacity to clone (use 2 or more axis for the same letter), there may be some output differences:  
- Depending of axis number, status reports can give from 3 to 6 axis position values,
- Axis parameters can be from $1x0 to $1x5,  
- The $I output report [VER:], [AXS:] and [OPT:]. the AXS sentence report the axis number and axis names. For example, [AXS:5:XYZAC] mean there is 5 axis with names respectively X, Y, Z, A and C. The reported values of axis will be in this order. The [OPT:] output have been extended with some new values.
- The axis names can be any combination of X, Y, Z, A, B, C, U, V, W, D, E & H. XYZUVW are for linear axis and ABCDEH for rotationals.  
  
@++;
Gauthier.

svenhb, 2021-11-20:
>The $I output report [VER:], [AXS:] and [OPT:].  [AXS:5:XYZAC] mean...

That's interesting - thanks.
The rest I already knew 🙂

amoineau, 2021-11-22:
Sven you're amazing, thanks for the changes !

closed 2022-02-15 by fra589