fra589/grbl-Mega-5X #230: ENDSTOPs not working

depeyrace, 2021-11-15:
Hello, I am trying to setup my CNC mill. Sadly I am having an issue with endstop that are not working.
So my questions are:
Can you check status of the pin via some GCODe command?
What pins should I use?(I used pinout from original GRBL)
Should I use any resistors? (Currently I am connecting it to ground)

And I didn’t edit any part of the source code.

fra589, 2021-11-15:
Hi,  

Reading the wiki is always a good begining...  

The ? command give the Input Pin State (see https://github.com/fra589/grbl-Mega-5X/wiki/grbl-Mega-5X-v1.2-interface)
The pinout is like the Grbl-Mega original one, not the original Grbl which run on Arduino uno, and not on Mega2560. The pinout is described in the wiki : https://github.com/fra589/grbl-Mega-5X/wiki/grbl-Mega-5X-pinout, no need to add resistors.  

@++;
Gauthier.

depeyrace, 2021-11-15:
Thanks for fast reply, I was using pinout from original UNO GRBL.  Now, that i have the right one it is working perfectly.

closed 2021-11-15 by depeyrace