fra589/grbl-Mega-5X #233: Digital or maybe even analogue inputs via the USB interface ?

GrblGru, 2022-01-03:
Hi Gauthier,

is it possible to get digital or maybe even analogue inputs from my Grbl interface 'GrblGru' via the USB interface ?
The background of the question is that some 5-axis robot arm users would like to create macros in which they would like to start functions depending on the current input states.

Again, thanks for your great work :)
GrblGru

fra589, 2022-01-04:
Hi @GrblGru,  
  
Yes, it should be possible...  
I plan to work on Grbl-mega-5X to add the extra PWM output functions by the end of this quarter, I will add this to the TODO list.  
Stay connected :-)  
  
@++;  
Gauthier.

GrblGru, 2022-01-04:
Hey, that's great. :)
Please let me know if I can help with this, e.g. with testing the interface.

Best regards
GrblGru

fra589, 2022-05-10:
Hi @GrblGru,  
  
With a tiny bit of delay :smiley:, yesterday I posted the version 1.2f which implements up to 4 optional digital inputs.
To use them, you need to enable `USE_DIGITAL_INPUT` in config.h.  
  
Then, the digital inputs status are visible in the "Accessory state" field (`A:SCFMD`) of the machine state in response to the command "`?`".  
[image]
  
Tell me if this is good for you, I continue to work on Analog inputs, but it's a little bit more complicated and have a risk to take a little more time.  
  
@++;  
Gauthier.

GrblGru, 2022-05-12:
Salut Gauthier,

thank you for thinking of me. And of course it's great that you created this possibility. :)
Unfortunately, I don't have much time at the moment either, but I will try it out as soon as possible and give you feedback.

Thanks again, also on behalf of the many users of your software.

Best regards
GrblGru

Ams-Laser, 2022-05-12:
I needed this function myself, Will do some testing and let you know

closed 2022-06-26 by fra589