fra589/grbl-Mega-5X #239: Cycle start button

arlimbad222, 2022-01-19:
Today I am trying to start cycling by pressing the cycle start button I use one nc push button for my application but somehow the button is not working .

I select a NC file from ugs and press the button but not working but when I start cycle from ugs then I press feed hold button it's working and then i press cycle start button it also working.

So I think it's not cycle start button it's just resume button.

Is there any way to start cycling by just pressing the button because it's hard to start the cycle all the time by keyboard or pc.

I have 100 same jobs at the time so that cycle start button help me a lot if possible then please someone give me an idea to do the same.

Ams-Laser, 2022-01-19:
Cycle start does not start the job its more like a Cycle resume button, i
would love if it coudl actually start the job

arlimbad222, 2022-01-20:
Hello @Ams-Laser  i think @fra589 has some solution about this things.

fra589, 2022-01-20:
Hi @arlimbad222,  
  
Grbl is just a peripheral like a printer or a touchscreen... It has no way of knowing that the software driving it has opened a file waiting to be printed or displayed.  
When the "cycle start" button is pressed, if Grbl is in "RUN" mode it places it in "HOLD", ie it stops all movements and blocks the arrival of new GCode commands. A new press of it return in normal RUN mode.  
The status of the button is sent to the control software in the status message "Input pin state" (`Pn:XYZPDHRS`). Search for "Input Pin State in " https://github.com/fra589/grbl-Mega-5X/wiki/grbl-Mega-5X-v1.2-interface.  
It is up to the control software to process this status to perform the cycle start when the machine is idle and there is an open file waiting for running.
It's in the [cn5X's](https://github.com/fra589/cn5X) TODO list but I haven't programmed it yet because I haven't needed it for my own machine yet and haven't been asked to :-)

@++;
Gauthier.

arlimbad222, 2022-01-20:
Hi @fra589 

Thanks for that (pn) idea I am using my own grbl sender so now I ask my devloper and make those changes.

Yes again thank you for making these grbl 5x 😊

closed 2022-01-20 by arlimbad222