fra589/grbl-Mega-5X #242: G2 G3 command from Fusion360

LinusDerome, 2022-01-23:
Hi!!

My machine is a LowriderV2 with two motors on Y & Z and one on X. All are connected to a single driver on a Ramps 1.4 so I have made the necessary changes in config for the axles to go right and to Auto Squaring.

Had huge problems at first with Homing when the duplicate motor refused to go home on Y, but after separating the endswitches cables and the motor cables it goes fine! :-)

But now to the problem I can´t solve.
Encounteres a strange problem when my machine gets G3 and G2 commands. this is given from Fusion 360 when you have to helix down.
When it gets these, motor 1 on the Y axis starts going linearly back and forth about 200mm in about 267mm feed rate for each command. and I mean only 1 moves the duplicated motor on axis 4 in config, it stands still.

when it has made these movements it continues as nothing has happened to the rest of the code but then completely out of square.

Have tried with Universal Gcode sender and OpenBuilds software, same problem on both.

Has in Fusion changed from Helix to Plunge and it helps in the initial phase, but once Fusion puts in a G3 G2 for some reason. its the same problem.

Can´t understand what´s going wrong, I guess it lies in how GRBL interprets the command or something like that but can not solve it myself.

so if there is something in this version of GRBL or if it´s Fusion that does it, I do not know, and can´t find out which.

I hade a DUO + CNC sheild before with standard GRBL installed, then I didn´t get these problems, so it´s new with my new set of hardware and software.

Thanks in advance!

RaphaelDives, 2022-01-23:
Hey @LinusDerome 
that’s sounds like wrong/not symmetrical axis parameters to me…  Anyway (as always) you should post your configuration ($$ output). And which axis (on the ramps board) you use for which motor (especially the two Y motors)…
Maybe also the changes you did in config.h and cpu_map.h (if any…).

Cheers Raphael

fra589, 2022-01-24:
Hi @LinusDerome,  
  
Like @RaphaelDives say, post your $$ output, and then also the $I output to know the exact version of grbl-Mega5X you have and the axis definitions.  
  
Reading your problem, it seems you made many test with different axis definitions...  
When the axis definition is changed (axis number, axis names, axis clonong, etc.) it very important to issue theGrbl's `$RST=*` command tho reset all parameters in the flash memory.  
Without this, some move results may be unpredictable!!!  
  
@++;  
Gauthier.

LinusDerome, 2022-01-24:
Hi @RaphaelDives and @fra589 

I would of course have included in the first post sorry.
but will suply it out next time I'm in the workshop, hopefully tonight.

But you do mean to reinstall Aurduinon with new config don´t help, you have to restore from eg OpenBuild as well?
May start by testing it of course.

fra589, 2022-01-24:
No, I'm not talking about restoring the firmware or reinstalling Arduino...  
  
When you change the number of axis or the axis names or the axis positions, you need to reset the parameters eeprom memory to "factory profile" (see Grbl command [$RST=*](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Commands)) because the number of parameters is not the same from the anterior version to the new one, so the eeprom mapping is not the same.
This generally produce unpredictable result.
  
To do this, send the Grbl's command: `$RST=*` with a terminal or if with your GCode sender it allow you to send any command.  
  
@++;  
Gauthier.

LinusDerome, 2022-01-25:
OMG!!! 
It was that easy!!! 
The recovery did good :-)

closed 2022-01-25 by fra589