fra589/grbl-Mega-5X #245: compiling error

Ramr93, 2022-01-26:
In file included from C:\Users\USER\Documents\Arduino\libraries\grbl/config.h:32:0,
                 from C:\Users\USER\Documents\Arduino\mega\mega.ino:1:
C:\Users\USER\Documents\Arduino\libraries\grbl/grbl.h:71:4: error: #error "Required HOMING_CYCLE_0 not defined."
   #error "Required HOMING_CYCLE_0 not defined."
    ^~~~~
C:\Users\USER\Documents\Arduino\libraries\grbl/grbl.h:99:4: error: #error "WCO refresh must be greater than one."
   #error "WCO refresh must be greater than one."
    ^~~~~
C:\Users\USER\Documents\Arduino\libraries\grbl/grbl.h:102:4: error: #error "Override refresh must be greater than zero."
   #error "Override refresh must be greater than zero."
    ^~~~~
exit status 1
Error compiling for board Arduino Mega or Mega 2560.

Ramr93, 2022-01-26:
please help to resolve this problem

fra589, 2022-01-26:
i @Ramr93,  
  
What compiler are you using to compile grbl?  
If it is with Arduino software, what is the version?  
Did you follow the wiki indicating the procedure: [Compiling-grbl-Mega-5X](https://github.com/fra589/grbl-Mega-5X/wiki/Compiling-grbl-Mega-5X)?  
Notably by checking that there is only one version of grbl in the libraries folders...  
Did you change settings in config.h, cpu_map.h, default.h, or anywhere?  
  
The errors you get mean that:  
- Either you have modified the values of REPORT_WCO_REFRESH_IDLE_COUNT and REPORT_OVR_REFRESH_IDLE_COUNT with invalid values and deleted all the definitions of HOMING_CYCLE in the config.h file,  
- either the compiler does not find the right config.h file in the right place...  
  
@++;  
Gauthier.

Ramr93, 2022-01-26:
hi thanks for your replay
 I'm using Arduino 1.8.19  compile and not touch any config files.
i simply followed the : Compiling-grbl-Mega-5X procedure but the error is reputing. 
 can you sagest with any video example

fra589, 2022-01-26:
You must not have understood the procedure well or your Arduino software or your Windows is not working correctly.  
https://user-images.githubusercontent.com/19529154/151146673-8816c3f9-96fd-40ae-9e65-3f8d012bb6b6.mp4
I work under Linux... It's the same procedure under Windows.  
  
@++;  
Gauthier.  
<< Now people don't know how to read anymore... They only know how to watch videos :worried: >>

fra589, 2022-01-26:
https://user-images.githubusercontent.com/19529154/151184282-e6ca2de4-fab7-4582-bbd7-dad559a50085.mov

Ramr93, 2022-01-27:
thank you ... install successfully..  thank you very much

closed 2022-01-27 by fra589