fra589/grbl-Mega-5X #249: G code doesnt run

PJAlves01, 2022-02-04:
Hi everyone!
I changed my controller from a UNO to a MEGA.
The ONLY thing i did was clone the X and Y Axis.
I can jog very well the machine on CNCjs.
But when i Run the g code (that worked on uno with grbl) it just doesnt work. No errors, nothing. Just doesnt move

Help please, i dont know what to do more.

fra589, 2022-02-04:
Hi @PJAlves01,  
  
grbl-Mega-5X work perfectly well for me and many peoples.  
Without your detailed configuration information ($$ result), exact grbl-Mega-5X version ($I result) and the incriminated GCode file, it will be very difficult to understand what happen to you...  
  
@++;  
Gauthier.

PJAlves01, 2022-02-04:
Thank you for your reply.
Neither G-code from Aspire or Fusion 360 works. (Grbl post processor)
In CNCjs it appears to be moving, but the steppers just turn off and dont move

PJAlves01, 2022-02-04:
[12.txt](https://github.com/fra589/grbl-Mega-5X/files/8003840/12.txt)

RaphaelDives, 2022-02-04:
Hey,
I maybe had a similar problem with Cambam. It would load the gcode, i could start it and then my bcnc said "done". I found out it was an M command in the fileheader... Since when I remove the header of my files, the whole code runs without error. (But maybe this is a bcnc error... ;) 
Cheers
Raphael

PS: Can you just try out some simple G Commands? Do they work?

PJAlves01, 2022-02-04:
That is from Aspire, which worked very well on Uno.
I really dont know what is happening (I did $RST=*)

PJAlves01, 2022-02-04:
> 

No M commands in here :/

fra589, 2022-02-04:
Hi @PJAlves01,  
  
Your config and GCode file seams to be good...  
Your file run normally on my machine with full moving...  
Are you shure your Grbl is not in "Check gcode mode" (result of $C command) ?  
Search on the CNCjs side or perhaps you can issue a test with another Grbl sender...  
  
@++;  
Gauthier.

PJAlves01, 2022-02-04:
Thank you for tacking your time to help me!

I really dont know what is happening. I have tried sending with UGS and the same thing happens. Its so strange.

That must not be the case, the Z Axis moves up and then everything shuts down (but the g code keeps running on CNCjs)

PJAlves01, 2022-02-04:
Next thing to try is using my other PC. Really getting out of options here :/

fra589, 2022-02-04:
Do you confirm that you use standard Arduino Mega with RAMPS shield ?
If this is not a RAMPS shield, perhaps you have to adapt the pinout...
https://github.com/fra589/grbl-Mega-5X/wiki/grbl-Mega-5X-pinout

PJAlves01, 2022-02-04:
Im not using Ramps but everything is connected correctly. (Im using TB6600 drivers) 
I can manually move the machine on CNCjs.

PJAlves01, 2022-02-04:
I had Stepper enable connected to spindle enable. Maybe i need glasses :/
Sorry to bother you.
Thank you for your time!

closed 2022-02-04 by PJAlves01