fra589/grbl-Mega-5X #252: Invert Spindle PWM

GatzeTech, 2022-02-16:
I don't know if this is the right place to ask this question, but is it possible to invert the spindle PWM (preferable without changing the firmware) ?

I managed to rebuild a Mach3 board to Grbl, but on the outputs there is a optopcoupler in between which inverts the signal.

The only serious thing I found is this:
[https://www.cnczone.com/forums/diy-cnc-router-table-machines/406528-grbl-mega-5x-ramps1-4-superpid.html](url)

fra589, 2022-02-17:
Hi @GatzeTech,  
  
I don't quite understand what you need.  
Can you be more specific in describing your problem?  
  
<<invert the spindle PWM>> does not make technical sense for me...  
A PWM signal is an electrical signal composed of +5V pulses whose pulse width is variable, which varies the effective voltage.  
Inverting it would mean generating -5V pulses? This is an electronic hardware problem, there is no software solution for it.  
  
@++;  
Gauthier.

GatzeTech, 2022-02-17:
Thank you for your answer.

I should be a bit more specific, this is not what I meant, I think you can better say, 'invert the dutcycle of the PWM 

What I need is, is that if for example the minimum spindle rpm is 1 and maximum is 1000 and if you sent for example M3 S250, the positive periode of the PWM should be 75% (instead of 25%).

And if you send M3 S750, the high period of the PWM should be 25% (instead of 75%)

This because I have a optocoupler between my output and pin of the microcontroller and inverts the signal.

But I have some experience with programming Avr's, I'm figuring it out right now, it's not that hard.

fra589, 2022-02-18:
Hi @GatzeTech,  
  
It was not hard to implement this new compilation config option. I just pushed a new v1.2e version which add this option in config.h.

To unable it, uncomment the line number 96 in the config.h file like this:
`#define INVERT_PWM_VALUES`

Tel me if it's that you need.

@++;  
Gauthier.

closed 2022-03-18 by fra589