fra589/grbl-Mega-5X #253: Serial.c --- volatile wrong way around on serial_rx indicies

casiobearing, 2022-02-22:
Just noticed "volatile" is on the wrong index for the serial_rx indicies....

uint8_t serial_rx_buffer_head = 0;
volatile uint8_t serial_rx_buffer_tail = 0;

it should be on serial_rx_head as that is modified within the serial RX ISR.
:o)

fra589, 2022-02-24:
Hi @casiobearing,  
  
Thanks for reporting! :smiley:  
  
You are right, but not only :-)  
In fact, the 3 variables: `serial_rx_buffer[]`, `serial_rx_buffer_head` and `serial_rx_buffer_tail` need to be volatile as there are all modified in ISR serial.c's interrupts.  
  
I just pushed the v1.2e build 20220224 which correct this.  
  
@++;  
Gauthier.

casiobearing, 2022-02-24:
Hiya,

No problem :o) I spotted it while stress testing serial comms for my project.
 I'm currently designing a remote control circuit board to sit inside my plasma cutter so it can report arc voltage (as well as switch on \ off plasma and set amperage), with the eventual plan to integrate a PID THC (torch height control) into GRBL.
I'm currently using a fork of Atmega2560 GRBL, using serial port 2 for the plasma cutter control comms.

The bit I haven't figured out yet is taking over the Z control from GRBL. I was thinking it could in real-time update the tool height instead of Z, but have not yet investigated that method.
The global chip shortage is causing me delays in getting the circuit board design finished but might have something usable in 2 months time.

I have a modified "sleep.c" module that supports multiple millisecond timers that you may be interested in for GRBL. It makes it easier to implement extra timers and has minimal impact on GRBL operations (the timer interrupt is never disabled).

Regards
ian

closed 2022-06-26 by fra589