fra589/grbl-Mega-5X #254: Wrong movements while running

LinusDerome, 2022-02-22:
Hey!

Now I have more problems unfortunately.
When I run my current project, the machine makes completely unexpected movements. See pictures.

[image]
[image]
[image]
[image]
[image]

The first picture is from my tests, it happens at about the same time, but there are variations, both on how the error forms, and where it starts.

here it is the code that is run in small band with the error.
**try 1**
>>> G1X311.884Y211.778
ok
>>> X312.222Y200.442
ok
>>> Y144.334
ok
>>> X117.999Y109.964
ok
>>> X90.441Y105.087Z-8.988F600
ok
>>> Y160.119Z-10.93
ok
>>> X91.333Y182.261Z-11.712
ok
>>> X92.621Y190.305Z-12
ok
>>> X94.498Y202.037F800
ok
ok
>>> G2X116.905Y241.36I73.64J-15.917
ok
>>> G1X124.678Y247.73
ok
>>> X133.812Y253.639
>>> X144.416Y259.086
ok
>>> X156.595Y264.07
ok
>>> X170.448Y268.588
ok
>>> X186.071Y272.637
>>> X203.547Y276.216
ok
>>> X218.408Y278.482
ok
>>> X231.986Y279.71
ok
>>> X244.315Y279.922
ok
>>> X255.431Y279.148
ok
>>> X265.367Y277.418
ok
ok
>>> G2X310.854Y222.246I-12.747J-56.846
ok
>>> G1X311.884Y211.778
ok
>>> X312.222Y200.442
ok
>>> Y144.334
ok
>>> X90.441Y105.087
>>> Y160.119
ok
>>> X91.333Y182.261

**try2**
ok
>>> X92.621Y190.305Z-12
ok
>>> X94.498Y202.037F800
ok
ok
>>> G2X116.905Y241.36I73.64J-15.917
ok
>>> G1X124.678Y247.73
ok
>>> X133.812Y253.639
>>> X144.416Y259.086
ok
>>> X156.595Y264.07
ok
>>> X170.448Y268.588
ok
>>> X186.071Y272.637
>>> X203.547Y276.216
ok
>>> X218.408Y278.482
ok
>>> X231.986Y279.71
ok
>>> X244.315Y279.922
ok
>>> X255.431Y279.148
ok
>>> X265.367Y277.418
ok
ok
>>> G2X310.854Y222.246I-12.747J-56.846
ok
>>> G1X311.884Y211.778
ok
>>> X312.222Y200.442
ok
>>> Y144.334
ok
>>> X90.441Y105.087
>>> Y160.119
ok
>>> X91.333Y182.261

Machine settings:

[19:59:31] [ ] Grbl 1.2d ['$' for help]

[19:59:31] [ ] [MSG:'$H'|'$X' to unlock]

[19:59:32] [ $$ ] $0=10 ;Step pulse time, microseconds

[19:59:32] [ $$ ] $1=255 ;Step idle delay, milliseconds

[19:59:32] [ $$ ] $2=0 ;Step pulse invert, mask

[19:59:32] [ $$ ] $3=0 ;Step direction invert, mask

[19:59:32] [ $$ ] $4=0 ;Invert step enable pin, boolean

[19:59:32] [ $$ ] $5=1 ;Invert limit pins, boolean

[19:59:32] [ $$ ] $6=0 ;Invert probe pin, boolean

[19:59:32] [ $$ ] $10=1 ;Status report options, mask

[19:59:32] [ $$ ] $11=0.020 ;Junction deviation, millimeters

[19:59:32] [ $$ ] $12=0.001 ;Arc tolerance, millimeters

[19:59:32] [ $$ ] $13=0 ;Report in inches, boolean

[19:59:32] [ $$ ] $20=0 ;Soft limits enable, boolean

[19:59:32] [ $$ ] $21=0 ;Hard limits enable, boolean

[19:59:32] [ $$ ] $22=1 ;Homing cycle enable, boolean

[19:59:32] [ $$ ] $23=10 ;Homing direction invert, mask

[19:59:32] [ $$ ] $24=300.000 ;Homing locate feed rate, mm/min

[19:59:32] [ $$ ] $25=1000.000 ;Homing search seek rate, mm/min

[19:59:32] [ $$ ] $26=250 ;Homing switch debounce delay, milliseconds

[19:59:32] [ $$ ] $27=15.000 ;Homing switch pull-off distance, millimeters

[19:59:32] [ $$ ] $30=12000 ;Maximum spindle speed, RPM

[19:59:32] [ $$ ] $31=550 ;Minimum spindle speed, RPM

[19:59:32] [ $$ ] $32=0 ;Laser-mode enable, boolean

[19:59:32] [ $$ ] $100=100.000 ;X-axis steps per millimeter

[19:59:32] [ $$ ] $101=100.000 ;Y-axis steps per millimeter

[19:59:32] [ $$ ] $102=400.000 ;Z-axis steps per millimeter

[19:59:32] [ $$ ] $103=100.000 ;undefined

[19:59:32] [ $$ ] $104=400.000 ;undefined

[19:59:32] [ $$ ] $110=5000.000 ;X-axis maximum rate, mm/min

[19:59:32] [ $$ ] $111=5000.000 ;Y-axis maximum rate, mm/min

[19:59:32] [ $$ ] $112=800.000 ;Z-axis maximum rate, mm/min

[19:59:32] [ $$ ] $113=5000.000 ;undefined

[19:59:32] [ $$ ] $114=800.000 ;undefined

[19:59:32] [ $$ ] $120=300.000 ;X-axis acceleration, mm/sec^2

[19:59:32] [ $$ ] $121=300.000 ;Y-axis acceleration, mm/sec^2

[19:59:32] [ $$ ] $122=200.000 ;Z-axis acceleration, mm/sec^2

[19:59:32] [ $$ ] $123=300.000 ;undefined

[19:59:32] [ $$ ] $124=200.000 ;undefined

[19:59:32] [ $$ ] $130=400.000 ;X-axis maximum travel, millimeters

[19:59:32] [ $$ ] $131=200.000 ;Y-axis maximum travel, millimeters

[19:59:32] [ $$ ] $132=200.000 ;Z-axis maximum travel, millimeters

[19:59:32] [ $$ ] $133=200.000 ;undefined

[19:59:32] [ $$ ] $134=200.000 ;undefined

LinusDerome, 2022-02-22:
(Komplett D)
(Machine)
(  vendor: Linus)
(  model: Lowrider cnc)
(  description: Lowrider cnc)
(T1  D=6 CR=0 - ZMIN=-15 - flat end mill)
G90 G94
G17
G21
(When using Fusion 360 for Personal Use, the feedrate of)
(rapid moves is reduced to match the feedrate of cutting)
(moves, which can increase machining time. Unrestricted rapid)
(moves are available with a Fusion 360 Subscription.)

(2D Adaptive5)
T1
S29000 M3
G54
M8
G0 X89.078 Y88.491
Z15
G1 Z5 F800
Z2.55
G2 X90.57 Y93.993 Z0.971 I0.746 J2.751 F333.3
X89.078 Y88.491 Z-0.608 I-0.746 J-2.751
X90.57 Y93.993 Z-2.186 I0.746 J2.751
X89.078 Y88.491 Z-3.765 I-0.746 J-2.751
X90.57 Y93.993 Z-5.344 I0.746 J2.751
X92.674 Y91.242 Z-6 I-0.746 J-2.751
X86.974 Y91.242 I-2.85 J0
X92.674 Y91.242 I2.85 J0
G1 X92.677 Y91.044 F800
X92.688 Y90.976
X92.704 Y90.735
X92.679 Y90.494
X92.614 Y90.262
X92.499 Y89.959
X92.455 Y89.858
X92.391 Y89.757
X92.113 Y89.355
X92.05 Y89.272
X91.964 Y89.189
X91.562 Y88.833
X91.499 Y88.787
X91.437 Y88.751
X90.957 Y88.499
X90.857 Y88.455
X90.756 Y88.427
X90.353 Y88.328
X90.152 Y88.288
X89.951 Y88.286
X89.548
X89.447 Y88.293
X89.346 Y88.315
X88.944 Y88.414
X88.742 Y88.473
X88.541 Y88.577
X88.21 Y88.751
X88.138 Y88.793
X87.937 Y88.964
X87.723 Y89.153
X87.615 Y89.254
X87.534 Y89.355
X87.256 Y89.757
X87.192 Y89.858
X87.149 Y89.959
X86.996 Y90.362
X86.929 Y90.563
X86.904 Y90.764
X86.855 Y91.167
[... 1112 lines cut ...]
G1 X46.625 Y247.439
X46.591 Y247.2
X44.521 Y226.606
X44.508 Y226.398
X43.825 Y204.245
X43.824 Y40
G3 X47.344 Y37.046 I3 J0
G1 X129.493 Y51.525
G2 X130.188 Y51.038 I0.104 J-0.591
G1 X130.292 Y50.448
X130.308 Y50.355 Z-14.993
X130.324 Y50.265 Z-14.971
X130.339 Y50.179 Z-14.935
X130.353 Y50.1 Z-14.885
X130.366 Y50.03 Z-14.824
X130.376 Y49.97 Z-14.753
X130.385 Y49.921 Z-14.672
X130.391 Y49.886 Z-14.585
X130.395 Y49.864 Z-14.494
X130.396 Y49.857 Z-14.4
Z7
Z12
M9
M5
M30

LinusDerome, 2022-02-23:
[19:22:23] [ GCODE Parser ] GCODE File Loaded, please wait while we render a preview...

[19:22:24] [ GCODE Parser ] GCODE Preview Rendered Succesfully: Total lines: 3647 / Estimated GCODE Run Time: 00h:21m

[19:22:31] [ $I ] [VER:1.2d.20210930:]

[19:22:31] [ $I ] [AXS:5:XYZYZ]

[19:22:31] [ $I ] [OPT:VNMGH,35,255,80]

[19:22:31] [ $I ] ok

fra589, 2022-02-24:
Hi @LinusDerome,  
  
This also may be due to loss of steps in the stepper motors...  
  
Your GCode file seams to be right. But the working speed is may be too speed for cutting forces.  
I think that your speed parameters and especially acceleration may be are too large!  
  
Perhaps you can try lowest values in the parameters and also the F values in the GCode. And if it work, try to find the optimum for you.  
  
@++;  
Gauthier.

closed 2022-06-26 by fra589