fra589/grbl-Mega-5X #257: PWM does not work

crkie, 2022-02-27:
Hello,
I have installed the latest version 1.2e on my board. With this the PWM output does not work.
Back to 1.1p it works fine again. Did I miss something? With both is SPINDLE_PWM_ON_D8.

Rudi

fra589, 2022-03-01:
Hi @crkie,  
  
What you say seems strange to me... I use 1.2e with SPINDLE_PWM_ON_D8 on my own CNC milling machine without problem.  
There is many update between 1.1p and 1.2e, so, I think you need to reset the flash configuration parameters with the `$RST=*` grbl's command to avoid unpredictable behavior.  
  
Another source of problem when upgrading the Grbl's version via the Arduino IDE is conflict with files of older and newer version. You need to be sure that you delete all older version files in the Arduino's library directory before installing the newest version.  
  
cf. https://github.com/fra589/grbl-Mega-5X/wiki/Compiling-grbl-Mega-5X:
[image]
  
@++;  
Gauthier.

crkie, 2022-03-02:
Hello Gauthier,
i have found the problem. I have to uncomment "#define INVERT_PWM_VALUES" to make it work. It that is so intentional that default is inverted? 

Greetings Rudi

fra589, 2022-03-03:
Hi @crkie,  
  
OUPS!  
It was an error when I committed the INVERT_PWM_VALUES functionality... Sorry.  
  
I pushed the right version few minutes ago.  
  
Now, it should work like it's writed.
```// PWM signal inversion:
// In case of particular electronics, it may be necessary to invert the values
// of the PWM signal of the spindle. For example, if the minimum spindle 
// rpm is 1 and maximum is 1000, M3S250 will output 75% instead of 25% and
// M3S750 will output 25% instead of 75%. Disabled by default
//#define INVERT_PWM_VALUES
```
Leave it commented to have the normal behavior.  
  
@++;  
Gauthier.

closed 2022-03-04 by fra589