fra589/grbl-Mega-5X #261: Wiring diagram errors?

ozzyman, 2022-03-12:
Is the diagram (https://raw.githubusercontent.com/fra589/grbl-Mega-5X/7a293a9ce03231f18a8e9d3011e1f0b145ba9644/doc/images/grbl-Mega-5X_Wiring.svg) up to date? 

Axis 5 Max endstop is shown as D59 (There is no D59), probably should be A5 (RAMPS1.4 Aux2 pin 3)

fra589, 2022-03-13:
Hi @ozzyman,  
  
As you can see on the RAMPS 1.4 diagram from reprap.org: https://reprap.org/mediawiki/images/c/ca/Arduinomega1-4connectors.png, the Aux2 pin 3 of the RAMPS board is either A3 or D59. A3 should be use when used in analogic mode and D59 when used in digital mode. But finaly, it's the same.  

In grbl-Mega-5X, it is used as digital input, so, I used the D59 name for it.  
  
Furthermore, we can see in the Arduino's software source code pins definition for the Mega board (<Arduino_source_PATH>/hardware/arduino/avr/cores/arduino/pins_arduino.h):  
`#define PIN_A5   (59)`
  
@++;  
Gauthier.

drf5n, 2022-04-13:
> As you can see on the RAMPS 1.4 diagram from reprap.org: https://reprap.org/mediawiki/images/c/ca/Arduinomega1-4connectors.png, the Aux2 pin 3 of the RAMPS board is either A3 or D59. A3 should be use when used in analogic mode and D59 when used in digital mode. But finaly, it's the same.

A5, not A3:

[image]

and 

https://github.com/arduino/ArduinoCore-avr/blob/master/variants/mega/pins_arduino.h#L51-L66

fra589, 2022-04-14:
Hi @drf5n,  
  
Oups! you're right, a mistake wroted too fast, it's really A5.

@++;
Gauthier.

fra589, 2022-04-14:
Hi @ozzyman,  
  
Yes, you're right. It would be more rigorous.
This is from gnea's original comments...
I will have to remember to correct these comments and the documentation during the next updates...  
  
@++;  
Gauthier.

closed 2022-06-26 by fra589