fra589/grbl-Mega-5X #271: Dual Y-Axis Motors (One is working where other is not)

rehanzfr, 2022-04-29:
I am new with grbl. I am using mks gen l v2.1 board. This board with marlin is perfectly fine with dual y axis motors. However, I want to use grbl-mega-5x. In this I could not get an idea that how we can run dual y-Axis motors. At this time, the 2nd motor is attached to E1 pins. Please guide.

RaphaelDives, 2022-04-29:
Hello @rehanzfr,
congrats that you finally made it to grbl-mega-5x. ;)

Since this project is build for the RAMPS Board - I'm not shure if you can use it out of the box - although it should be possible for shure... :)

The first thing would be: https://github.com/fra589/grbl-Mega-5X/wiki/Compiling-grbl-Mega-5X

Second: cloning of axes is done at [config.h](https://github.com/fra589/grbl-Mega-5X/blob/edge/grbl/config.h) (right at the beginning), all the stuff you need to know is there in the comments!

As you can find here: https://github.com/fra589/grbl-Mega-5X/wiki/grbl-Mega-5X-pinout#grbl-mega-5x-pinout-on-the-ramps-14 the E1 is defined for the 4th axis (B). So if you have no use for the A axis, I would recommend to use the E0 port instead (I don'd think its possible to use axis 1,2,3 and 5). Of course you could also clone to both E Ports if you don't need them, or you can change the output for the 4th axis in the [cpu_map.h](https://github.com/fra589/grbl-Mega-5X/blob/edge/grbl/cpu_map.h) you will find it also at the beginning. Here is also the place to check wether the outputs are correct for your MKS board! :)

So, if you run into problems, your welcome to report!!

Cheers
Raphael

PS: The question about cloning axis is one of the most asked questions here, by the way... some searching in issues would have done the job as well. But hey I'm a nice person today... :D

RaphaelDives, 2022-04-29:
Damn, to slow... but at least you found it yourself! 😃 
I think you are on the right way, although imho it would be cleaner to use only 4 axis (instead of 5), in the case you don't need the A axis... (but I don't know about your setup...)

Don't forget the correct configuration for the 5th axis: $101=$104, $111=$114 and so on...
Good luck! :)

rehanzfr, 2022-04-29:
Thank you very much.

fra589, 2022-05-04:
Hi @rehanzfr,  
  
I'm glad that you found by yourself, but you may have little problems with it...
With your need to have 3 axis with cloned Y, this mean that there will be 4 linears axis.
In your solution, there is also the A axis which is a rotational axis.
So, the axis number definition should be :
```
#define N_AXIS 5        // Number of axes (3 to 6)
#define N_AXIS_LINEAR 4 // Number of linear axes, must be <= N_AXIS
```

The N_AXIS_LINEAR is internally used to know axes which need the millimeters to inch conversion for G20/G21 modes... With your solution, may be you will have problems if you need to work inches.

But warning ! The linear axes numbers must appears before rotational axes numbers. So, you have 2 solutions for your configuration:
- Connect your Y cloned axis to E0 ans the A axis to E1 (hardware change),
- Invert the axes 4 and 5 definitions in the cpu_map.h file (software change),

And define the axis name like this:
```
#define AXIS_1_NAME 'X'
#define AXIS_2_NAME 'Y'
#define AXIS_3_NAME 'Z'
#define AXIS_4_NAME 'Y'
#define AXIS_5_NAME 'A'
```

@++;  
Gauthier.

rehanzfr, 2022-05-04:
Yes, you are right. I had the problems even after what I defined above. I will try the new configurations you suggested and will report it back. Thank you very much.