fra589/grbl-Mega-5X #272: Changing x-max (limit) pin as a-min (limit) pin

futurau, 2022-05-17:
Hİ
I am not using max limit pins.
I want to use x-max (limit) pin as a-min (limit) pin,
and y-max (limit) pin as b-min (limit) pin.
Any help will be grateful.

fra589, 2022-05-17:
Hello @futurau,  
  
The documentation about remapping Grbl IO pins is in the Wiki:  
https://github.com/fra589/grbl-Mega-5X/wiki/Pinout-mapping-in-cpu_map.h  
  
@++;  
Gauthier.

closed 2022-10-20 by fra589

ariefadha, 2023-01-11:
In Wiki only shows how to remapping the pin, what should I do if I want to discard/delete x-max limit pin? because the x-max limit pin already used by a-min limit, and I don't need grbl to use x-max limit in software.

So a-min (limit) pin move to x-max (limit) pin.
And delete x-max defined by grbl. 

Should I comment out #define port and bit x-max?

RaphaelDives, 2023-01-11:
I would set the port and pin of x-Max to the same as the x-min.
Grbl defines if it is looking for max or min itself.
Or of course you can set it to any unused pin if there are any…
Cheers!