fra589/grbl-Mega-5X #279: motors off

wbendick, 2022-07-28:
When I disconnect from serial and then reconnect, my steppers lose power and my gantry crashed into my table or workpiece. I have $1 set to 255. Is there a workaround for this?

fra589, 2022-07-28:
Hello @wbendick,  
  
By design, the Arduino is reset by opening the serial port (auto reset) and/or on power up. So, the stepper power lost is the normal behavior.
  
Solution to disable this can be found here:  
https://playground.arduino.cc/Main/DisablingAutoResetOnSerialConnection/  
  
@++;  
Gauthier.

wbendick, 2022-07-29:
I see. Would it be possible to have the motors immediately power up after reset to prevent drops?

fra589, 2022-09-17:
Hi @wbendick,  
  
Perhaps you ca try to add a call to `st_wake_up();` in main.c...  
Add this on line 525 (just before the `report_init_message();` line.  
  
I don't verify if this work, but if no, this might be complicated to do.  
  
@++;  
Gauthier.

closed 2022-10-20 by fra589