fra589/grbl-Mega-5X #282: Digital Inputs add M66 "Wait on Input" GCode

fra589, 2022-08-06:
### Discussed in https://github.com/fra589/grbl-Mega-5X/discussions/280

<div type='discussions-op-text'>

<sup>Originally posted by **mcinquino** July 29, 2022</sup>
Hello,

I see that in the pinout diagram there are 4 digital inputs on pins 27,29,31 and 33 that are shown. I see that we have M codes -  M62, M63, M64, M65 to set the outputs that are shown above these inputs. In linuxCNC we would use M66 to wait on an input. I don't see that M66 is implemented in the list of G codes. Is there a way to use these inputs in G code? 

[image]

Thanks,
Mike </div>

fra589, 2022-08-06:
M66 is not yet implemented since nobody have asked it for now :)
Normally, it will be not so difficult to add M66 to the GCode interpreter...

Considering all my occupations, I could probably look into the matter after mid-October...

@++;  
Gauthier.

zalexzperez, 2022-09-11:
@fra589 
So, as I understand, those digital inputs can't be used right now?

I'd like to have a physical button press activate the homing sequence but I don't see a way to achieve this without digital inputs.

zalexzperez, 2022-10-02:
@fra589 I was actually thinking of being able to home the machine independently from the G-code sender program, as being able to turn on the machine and immediately pressing a button to home it would be more convenient (easier and time-saving).

I'm not seeing how my controlling programs (UGS and gSender, I prefer the latter) are able to react to (output status) accessory state changes. I barely managed to enable the visualization of output status messages on UGS by digging into the options and finding the "Show verbose output" option. 
Are you sure these programs can be configured to send commands on reading the output status?

Thanks in advance, 
Alejandro.

fra589, 2022-10-04:
Hi @zalexzperez,  
  
Since [gsender](https://github.com/Sienci-Labs/gsender) and [UGS](https://github.com/winder/Universal-G-Code-Sender) are open source, you can make your contribution by adding the necessary code...  
  
It's also possible to add code directly in Grbl by adding an input pin which react with homing or to performs the homing automatically when resetting Grbl (it's not too dificult do do).  
  
@++;  
Gauthier.

Ams-Laser, 2023-03-11:
Hello, is there hope on implementing M66 wait for input, this would be a real use for my cnc embroidery machine as to wait for neddle up before sending movement

Thanks

Ed

fra589, 2023-03-11:
Hi @Ams-Laser,  
  
It's still on the TODO list...
I'm very busy at the moment, I'm going to raise the priority, but it's going to take time... Be patient!  
  
@++;
Gauthier.

Ams-Laser, 2023-03-12:
Thanks, ill hold on 😉

M12b1t83i, 2023-03-20:
I need 259

Dprocess at 4 Val 

Max 9D

fra589, 2023-03-20:
@M12b1t83i  ????