fra589/grbl-Mega-5X #285: mega5x hotwire

Loko56, 2022-08-18:
@fra589 
Bonjour Gauthier,
je me permet de vous contacter pour un souci avec GRBL version fil chaud MEGA 5X de Keith Howlette (rcKeith)
Un bug d'affichage se produit sur le premier onglet(interface) qui affiche "décalage longueur" par dessus la fenêtre de l'axe U, rendant la lecture de la position de l'axe U  illisible.
La solution de Keith (repasser windows en anglais) est un peu contraignante.
Auriez-vous le temps et l'envie de vous pencher sur ce problème pour les utilisateurs français qui, comme moi, n'ont pas les compétences nécessaires pour le solutionner!
D'avance merci !
Cordialement.
Franck
ps: ci dessous la réponse de Keith
Hi Franck
A guy contacted me a while ago about this. I think its to do with the way windows is translating English to French. Are you able to switch to English I think this fixed it,
Keith

Loko56, 2022-08-25:
Bonjour Gauthier,

merci beaucoup d'avoir pris la peine de me répondre.

Je te joins les deux écrans demandés.

Le problème est que l'installation de GRBL5X Hotwire avec Windows (10) en français provoque un bug d'affichage, une surimpression de texte qui masque  la fenêtre d'affichage de l'axe U.

Même bug sous Win7.

En allant dans les paramètres  et en passant Windows en langue anglaise, GRBL5X Hotwire passe aussi en anglais et le bug d'affichage disparait...

Et il n'y a pas non plus d'option pour installer directement GRBL5X Hotwire en anglais( ce qui me conviendrait aussi), autre que de modifier Windows, ce qui est quand même contraignant.

Merci en tout cas de t'intéresser au problème, qui va surement te paraître banal !

Je te joins le lien du site de Keith pour info:

https://rckeith.co.uk/grbl-hotwire-mega-5x-for-cnc-foam-cutters/

Dans l'attente de tes bons conseils !

Merci d'avance.

Franck
[image]
[image]
k

fra589, 2022-08-26:
Bonjour Franck,  
  
Je pense que c'est dû à un problème dans le fichier de ressources de traduction française.
Il semble que le "U" soit traduit par "Décalage longueur" au lieu de rester "U".

On peut passer outre ce problème tout en gardant Windows en Français en supprimant les ressources de traduction. Windows restera en Français, et le logiciel en Anglais car il ne trouveras pas la version des ressources Française.  

Pour faire cela, tu cherche le dossier dans lequel est installé HotWire dans %APPDATA%\Local\Apps\2.0\...
Chez moi, je suis sous Linux avec Wine, j'ai des noms de répertoires bien compliqués (81OZ6CZE.PR0/7DA3LTVT.202/grbl..tion_0000000000000000_0005.0001_471eef88edb90f52)

Dans le dossier ou est installé le programme, il y a un dossier fr qui contient les fichiers suivants :  
```
Grbl HotWire Mega 5X 5.11.resources.manifest
Grbl HotWire Mega 5X 5.11.resources.dll
Grbl HotWire Mega 5X 5.11.resources.cdf-ms
```
Tu supprime ou tu renomme le dossier "fr", puis tu relance HotWire qui sera alors toujours en Anglais.

@++;  
Gauthier.

Loko56, 2022-08-26:
Bonjour Gauthier,
j'ai suivi tes instructions et ça fonctionne ! Le logiciel est en anglais et plus de bug d'affichage.
J'avais tenté de supprimer ce fichier FR du ZIP avant de faire l'install, mais ça bloquait l'installation et celle -ci ne se faisait donc pas.  Merci beaucoup pour ton aide rapide et efficace ! 
Je vais faire remonter l'information a Keith pour qu'il fasse le nécessaire.
Merci encore !
Franck

Loko56, 2022-08-27:
Bonjour Gauthier,
pour information, voici la réponse de Keith:

_Hi Franck
Many thanks for solving the problem with Gauthier. I'll amend the installation so it doesn't include that. As you may be aware it's based on GRB Panel and so that's where it came from. I was able to understand the dialogue you had with Gauthier on github thanks to Google's translation. 
I'll let you know when I have fixed the installation and upload it to the website. 
Have a good weekend 
Keith_

@++;
Franck

closed 2022-08-27 by fra589