fra589/grbl-Mega-5X #293: Auto gantry squaring

muhammadelmogy, 2022-09-21:
I have a cnc with 3 axes but with Y axis cloned like Shapeoko style machine

I am now have only 3 limit switches, one for every axis and I want to add another limit switch for the cloned Y to be able to make an auto squaring for the Y axis.
Is it possible to do this? if so what do I need to change?

fra589, 2022-09-23:
Yes, it is possible to do it,
Search Google for [grbl-mega-5x auto square](https://www.google.com/search?q=grbl-mega-5x+auto+square&client=firefox-b-d&ei=NMQtY66lLdmHxc8P_pGP0Ao&ved=0ahUKEwiuj_e4kav6AhXZQ_EDHf7IA6oQ4dUDCA0&uact=5&oq=grbl-mega-5x+auto+square&gs_lcp=Cgdnd3Mtd2l6EAMyBAgAEB46BwgAEB4QsANKBAhBGAFKBAhGGABQgxdYxypg5UBoAXAAeACAAX2IAdIBkgEDMS4xmAEAoAEByAEBwAEB&sclient=gws-wiz)  
And if you need specific help when triyng, come back with details of your config ($I, $$, config.h adjusts) and a real specific question.

closed 2022-09-23 by fra589

muhammadelmogy, 2022-09-23:
Instead of your useless answer, A simple answer to my question would be to simply tell what setting to change and that's it!
I already did my search before writing my question and could not find a definitive answer to that question!!

I went through at least half a dozen of pages in the issues and several forum pages trying to find how to do it, at the end after I Wrote this question I saw the todo file with the thing that I was asking for!!!
I was going to close the issue since the feature is in the TODO list but I said to my self there might be some work around and waited.
I will change to grblHAL with in a month so it was really waste of time to ask this question to you!!
I really hate to ask questions because of those type of arrogant people out there !!

langwadt, 2022-09-23:
@muhammadelmogy you'll get everything you paid for the software and service back ...

fra589, 2022-09-24:
Hello @muhammadelmogy,  
   
A minimum of courtesy and politeness (hello, please, thank you, cordially, ...) as well as specific questions generally make me more inclined to develop my answers.  
   
When I realized that Sonny Jeon no longer wanted to evolve Grbl, I decided to maintain and evolve it by creating grbl-Mega-5X with the adaptations and corrections I made for my own needs.  
I provide this maintenance completely free of charge on my free time.  
I add features that are requested of me when, after discussion, it appears to me that they are useful for several users and that they can be implemented without affecting the overall performance of Grbl.  
  
Apart from what I added at the start (going from 3 to 5 axes, then 6 axes), most of the new features I added are not useful to me and I don't need them(I don't have cloned axis on my milling machine, I don't need autosquaring, ...). I did it completely voluntarily because I thought it had an interest for the future of Grbl.  
   
Several development requests are in the TODO list, and by looking for a minimum in the [Issues](https://github.com/fra589/grbl-Mega-5X/issues) and in the [Discussions](https://github.com/fra589/grbl-Mega-5X/discussions), we can see what will be the future of Grbl.  
   
You say:  
> I already did my search before writing my question and could not find a definitive answer to that question!!  

You didn't say that in your original post, I can't guess!  You also didn't say either what you have found that didn't suit you.
   
You probably haven't searched well... I've researched auto squaring myself and found several projects on Github that deal with the subject. I did this research for inspiration as the work is in progress.
  
By giving you a search link, I didn't feel arrogant! I only sought to help by wasting the minimum of time. I prefer to use the little time I have for coding grbl-Mega-5X.
   
Good luck with grblHAL, because with the kind of message you write, you are not welcome here.
  
Cordially,
@--;
Gauthier.

muhammadelmogy, 2022-09-24:
If you do not want to answer then it is fine, leave the topic to someone else to help people and do not waste your time but
telling people that they do not ask a real question and tell them to search for the answer some where else while the answer could be just one or two lines if not called arrogant so what it is?!
Also, This is suppose to be a technical discussion so _(hello, please, thank you, cordially, ...)_ is just something people do not add usually in the technical discussion !!!!

I can thank you for your contribution, your work already helped me, but the way you answered me makes me feel not to!

You know, I have already modified your work in the past because some bugs in the homing code and I did not ask any questions as I remember because I made my research and was able to do what I want!
I do not have to mention I made my research!, what is the point of mentioning I made a research or not!! I never ask a question anywhere until I made a good a mount of research because I know people is not paid to set and answer me or I do not own there time, for that reason answering a question can take a lot of time which is something I usually do not want to waste!

> Good luck with grblHAL, because with the kind of message you write, you are not welcome here.

Man, this is not your house to tell people they are welcome or not, I will come and go when ever I want, it is an open source community not your house!

langwadt, 2022-09-24:
> Man, this is not your house to tell people they are welcome or not, I will come and go when ever I want, it is an open source community not your house!

if you paid attention to his username you'd see that it is his repository, so it is his house ...

RaphaelDives, 2022-09-24:
Hello @muhammadelmogy,
although I share Gauthier's Comment and I think he's absolutelyright...
What answer do you expect on your "question"?

> I have a cnc with 3 axes but with Y axis cloned like Shapeoko style machine

Congrats! :)

> I am now have only 3 limit switches, one for every axis and I want to add another limit switch for the cloned Y to be able to make an auto squaring for the Y axis. Is it possible to do this?

Yes, it is possible.

> if so what do I need to change?

Really? Come on... I did this myself a long time ago for my MPCNC (with two cloned axis and four limit switches...

So IF one would search for this you COULD find:
https://github.com/fra589/grbl-Mega-5X/wiki/Quick-Guide-to-Setting-Up-Your-Machine-for-the-First-Time
or maybe this:
https://github.com/fra589/grbl-Mega-5X/issues?q=clone
and after some reading this:
https://github.com/fra589/grbl-Mega-5X/issues/4
By the way, the topic comes up every month or so, there are many issues dealing with this...

Long story short:
1) Install your limit switch
2) Connect the wires
3) if necessary update the cpu_map.h
4) update the conf.h for cloned axis
5) update the homing cycle for homing the two axes simultaneously
6) upload it an have fun with a very good piece of software :)

(By the way IF one would look into the source code, it is very well commented and there are many clues for it, at least one could look into the code itself to see how stuff is done... but what do I know...)

And now for my opinion... There are many people out there that use this great peace of software, maintained by Gauthier (in his spare free time) and at least I'm very glad that he eventually find time for it and there are updates once in a while. Also there are people, willing to help, IF one writes a polite and proper question.
(In example: Hello, my name is xx, i have this hardware yy with that board zz, already tried this or that and I have a problem, that I cant's solve. I did changes abc to file 1,2,3 and this is my configuration $$. PLEASE help... Is this really so hard?!)
 
To be honest, I read your first question and thought about writing an answer, but since it would have been very similar to Gauthiers answer, because I found it quite unnecessary to write a answer to an already often and well anserwed topic...

And for the stuff with the open source community... no offense but a little bit of kindness does not hurt anyone. ;)
Sorry for the long text, but I had to write this...

Good look with whatever software you choose in the end, but you definitly should give the grbl-mega-5x a chance! :)
And of course I will give you advice and will try to help as good as I can, when you encounter problems!
(And I'm shure others will also...)

I whish You all a really nice evening! :)
Cheers,
Raphael

muhammadelmogy, 2022-09-24:
Believe it or not but when I build this machine I barely looked at any docs. and most of the time I just modified the code to do what I need and I did not ask as I remember anyone for anything! but I could not find a direct way to make the auto leveling, even I have a work around in my mind even before asking this question but I just wanted to make sure if there was a standard procedure or not!
Anyway, I have been using grbl-mega-5x for a long time really but I have lost interest in doing this anymore , but as I have mentioned I will be shortly converting to GrblHal since all original Grbl and its descendants are very limited when it comes to higher step rates, thanks!

fra589, 2022-09-25:
Hi @langwadt, hi @RaphaelDives,  
  
Many thanks to your support and attitude.  
  
The attitude of an ill-mannered moron such as @muhammadelmogy, who comes to insult me at my house _(those pages are those of a project that **I created** and that **I maintain** in a altruistic way)_ does not deserve that we waste time helping him or making him understand his stupidity.  
  
A technical discussion does not dispense with being polite and courteous! During a 40-year career in almost all IT professions, I had many technical discussions, always with politeness between well-bred people.  
  
If he had really found and fixed bugs in grbl-Mega-5X as he says, contributing to the project would have been the right attitude in the open source community. But I doubt that, having forked this project just 4 days ago, and that on his fork, there is no modification on his part, he really did what he wrote.  
  
In any case, considering all this inconvenience, he just win to be the first on the blacklist of my Github projects.  
  
@++;  
Gauthier.

Host1992, 2022-10-21:
[image]