fra589/grbl-Mega-5X #300: Hominng problem

enddDD, 2022-10-11:
Hi, i have a problem and dont understand why, i have limit switches on min and max, i have double y axis, when jogging everything is ok, the problem when homing is that one y (the y2 one)motor doesnt move in any direction but just freeze in its position and the other one basically tries to force it to move but it doesnt. What can be the problem? Thanks!

enddDD, 2022-10-11:
Do i have to uncomment the 4th axis? Am i right?

RaphaelDives, 2022-10-11:
> #if N-AXIS == 4 // 4 axis : homing #define HOMING_CYCLE_0 (1<<AXIS_3) . #define HOMING_CYCLE_1 ((1<<AXIS_1)|(1<<AXIS_2))) I have this. Do i need to comment/uncomment something? Thanks

That depends, on your axis definition (right at the beginning of config.h line 55-82).
Normally I would guess you have X/Y/YY/Z = 4 linear axis: 
It should be like #define N_AXIS 4
then it depends on your config:
e.g. 1=X 2=Y 3=Z 4=YY (YY=cloned Y axis)
then you have to define homing for both simultaneously:
#define HOMING_CYCLE_1 ((1<<AXIS_2)|(1<<AXIS_4)))
for both Y/YY axis

enddDD, 2022-10-11:
Yes i have x, y, y1 and z. It is a cnc router. Thank you for your information.

enddDD, 2022-10-11:
Ok, i modified, it looks alright i have changed so it homes z after x and only then the y(individual axis homing)but having another problem now. As i have on y only 2 limit switches- min and max, and are connected on ramps at ymin and ymax, now the problem is when the limit on y is touched, one motor stops, the other keep running, basically searching for the limit i guess. My y2 is at E0 on ramps board if it matters. What to do?

RaphaelDives, 2022-10-11:
Hey, thats not a (very) big deal!
You can use the same limit switch for more than one axis! 
(Although I would recommend individual switches -> autosquaring!)

Look into cpu_map.h and search for the MIN_LIMIT_PORT/MIN_LIMIT_BIT and MAX as well. (Starting at line 140)
Normally E0 should be the 4th axis,
so change the MIN_LIMIT_PORT_3 to the same as MIN_LIMIT_PORT_1, also for the BIT and for MAX_PORT/BIT.
So both axis will use the same pins and stop both motors! 😎

enddDD, 2022-10-11:
Wow, thank you verry much for your help. I have done the modifications and now works perfectly. Thanks again i couldnt done it without your help!

closed 2022-10-11 by enddDD

RaphaelDives, 2022-10-11:
Your welcome! 😎